alf45tar/PedalinoMini #475: Send CC Sequence when specific PC Command is received

wcraffonara, 2023-11-13:
Hi,
  Do I understand correctly that it is currently not possible to define an action that generates a defined CC Command Sequence at the MIDI OUT when a specific PC Command is received from the MIDI IN interface? 

Best regards
Walter

alf45tar, 2023-11-16:
It is not possible. Can you explain better your scenario?

wcraffonara, 2023-11-16:
Ok. Here is my scenario:
I have a BOSS ES-8 with which I control several MIDI devices. One of them is a TC Electronic Plethora X3 where different presets can be recalled via PC command. CC commands can then be used to adjust the parameters of the respective presets. To call up a preset and adapt certain parameters, the following commands must be sent, for example:
PC CH 1 2 + CC CH1 103 0 + CC CH1 104 1 + CC CH1 105 60

The problem now (and this has already been confirmed by BOSS and TC Electronic) is that the two devices use different implementations of the MIDI protocol. The BOSS control unit sends such MIDI sequences as "BULK commands", while TC Electronic expects individual commands, i.e. PC CH1 2, then CC CH1 103 0, then CC CH1 104 1 and finally a CC CH1 105 60, which also expects a slight delay of a few ms between the commands. BOSS, on the other hand, sends the entire sequence as a "BULK command" and therefore the TC Electronic only executes the first MIDI command (PC command) and ignores the CC commands.

The proposed solution from BOSS/TC Electronic is to send individual commands and my idea would have been to configure a PedalinoMini so that it recognizes a PC CH20 1, for example, and then generates a sequence of individual PC and CC commands at the MIDI OUT with a few ms delay between the individual commands. I already have the circuit diagram:
  - PedalinoMini has a MIDI IN and MIDI OUT interface and there is also a MIDI THRU interface which is realized in hardware
  - MIDI IN of PedalinoMini is connected with BOSS ES-8
  - MIDI OUT of PedalinoMini is connected with TC Electronic Plethora X3
  - MIDI THRU is connected with the rest of my devices

Or do you have a better idea how a translator between PC and CC commands could be realized?

Regards
Walter

alf45tar, 2023-11-22:
Why you are not connecting an external control port of BOSS ES-8 to PedalinoMini?

wcraffonara, 2023-11-23:
Yes, that could possibly work. I would then have to configure the Boss ES-8 so that when I press a button it first sends the PC 1 2 command (to select the bank on the Plethora X3) and then the ES-8 would have to activate an external control port to tell the PedalinoMini that a further sequence with e.g. CC CH1 103 0 + CC CH1 104 1 + CC CH1 105 60 should be sent. 

However, as I no longer have any free external controls on the ES-8, I have now simply adapted the PedalinoMini.cpp and the whole thing works perfectly:
in the loop() loop I added the following lines (yes, some values are hard coded, I will change that):

  if (interfaces[PED_DINMIDI].midiIn && DIN_MIDI.read()) {
    DPRINTMIDI("DIN MIDI", DIN_MIDI.getType(), DIN_MIDI.getChannel(), DIN_MIDI.getData1(), DIN_MIDI.getData2());
    
    // check if Plethora X3 channel was received
    if (DIN_MIDI.getChannel() == 1) {
      // read CC command -> expect a CC 15 command to select the sequence to be send
      for(i=0; i<10; i++) {
        if (interfaces[PED_DINMIDI].midiIn && DIN_MIDI.read()) {
          // re-check the channel -> CH 1 for Plethora X3
          if (DIN_MIDI.getChannel() == 1) {
            // CC 15 needed
            number = DIN_MIDI.getData1();
            if(number == 15) {
              value = DIN_MIDI.getData2();            
              DPRINT("Send Sequence to Plethora X3 -> CC 0x%02X   Value 0x%02X \n", number, value);
              midi_send(PED_SEQUENCE,0,0,value,true,0,0,0,0,0);
            }
          }
          break;    
        }
        delay(10); // wait 10ms
      }
    } 
  }

Regards
Walter