alf45tar/PedalinoMini #477: LILYGO T-Display-S3 On-Device Button functionality

Pedalnetics, 2023-12-02:
I just programmed a LILYGO T-Display-S3 via the sponsors repo (if this is not the right place for issues, please redirect me), and it appears the on-device buttons are not functioning properly/as documented.

The boot button does not do anything after bootup; during bootup,it does properly skip steps like Wifi/Smart/etc; I can also hold it on-power-up to boot the special modes.

Once the device is booted and displaying the bank screen, the boot button no longer does anything. There is another on-device button the docs call IO14; this does move forward profiles with a single click, but none of the other functions work: double-press to go back a profile, or long-hold to change/scroll through the display pages. (The RST button does reset the device, and the boot button does nothing at this point).

alf45tar, 2023-12-02:
Yes, it is the right place for issues.

T-Display-S3 has no default configuration for the moment because it is in early stage of development. You have to created yourself the Actions.

Pedalnetics, 2023-12-02:
[image]
There does seem to be a global action already defined for this (I didn't add these), so maybe it was commited by accident/wasn't fully working? If I change what Press does, it responds as-expected (i.e. if I change Press to Device Info, it does change the display to device info).
However neither Double Press or Long Press seem to work

jaylfc, 2023-12-15:
Could you elaborate a little on how to setup the buttons, or best to just wait for an update? Cheers!

alf45tar, 2023-12-15:
[image]

jaylfc, 2023-12-20:
Thanks, I'll give it a try :)

alf45tar, 2023-12-23:
Added same default config of T-Display for T-Display-S3 in 3.2.0

closed 2023-12-23 by alf45tar