alf45tar/PedalinoMini #478: LILYGO T-Display-S3 Not fully clearing screen after Wifi connection

Pedalnetics, 2023-12-13:
[image]

After setting up a Wifi connection and successfully connecting, and the device loads into a bank view where the bank name does not load anything into these pixels, you can still see remnants of the words "Connecting to" above the bank area.

In this config, the bank name is ":HXStomp"

alf45tar, 2023-12-13:
It will be fixed in the next version.

zavorra, 2023-12-23:
> It will be fixed in the next version.

I have also noticed that on lilygo t-display S3 the action tags are not properly displayed. Out of 8 actions only labels for 1 and 3 are (partially) shown

[image]

Is this part of the same incomplete implementation or am I doing something wrong with my conf?

alf45tar, 2023-12-23:
It is incomplete as of today.

zavorra, 2023-12-24:
If you want I could help on this.