alf45tar/PedalinoMini #479: LILYGO T-DISPLAY S3 cannot build

zavorra, 2023-12-13:
Hi alf45tar, 

It seems that I am not able to build you (impressive) project, cloned from the sponsor repo, neither on vscode nor on pio command line

I am trying to build for lilygo-t-display-s3, and i get this:
```
Compiling .pio/build/lilygo-t-display-s3/src/PedalinoMini.cpp.o
In file included from .pio/libdeps/lilygo-t-display-s3/Adafruit TinyUSB Library/src/Adafruit_TinyUSB.h:37,
                 from src/USBMidiOut.h:16,
                 from src/PedalinoMini.cpp:48:
.pio/libdeps/lilygo-t-display-s3/Adafruit TinyUSB Library/src/arduino/ports/esp32/tusb_config_esp32.h:49:2: error: #error "ESP32 Arduino core version 2.0.8 or later is required"
 #error "ESP32 Arduino core version 2.0.8 or later is required"
```

If I try to force version 2.0.8 on platformio.ini with
`framework       = arduino@2.0.8`
I get this error:
`Error: This board doesn't support arduino@^2.0.8 framework!`

On the other hand, if I force an older version of TinyUSB (older than 2.0.0, it seems) with 
```
lib_deps	= 	${common.lib_deps}
				adafruit/Adafruit TinyUSB Library@^1.9.0								;	https://github.com/adafruit/Adafruit_TinyUSB_Arduino

```
I am able to build and upload both firmware and file system

Is this fine/expected/ok for you?

Thank you in advance

Zavorra

alf45tar, 2023-12-13:
Check here https://community.platformio.org/t/error-esp32-arduino-core-version-2-0-8-or-later-is-required/36944

zavorra, 2023-12-13:
Ok, it is working. The actual problem was the 
`platform  	= espressif32@^6.0.0`
line in platformio.ini

Turning it into 
`platform  	= espressif32`
allowed platformio to install  espressif32 v. 6.4, which is based on arduino core 2.0.11

You may want to fix this

zavorra, 2023-12-13:
Changed the issue name, because  after successfully build and upload the firmware, and being able to configure wifi and some actions I noticed that USB_MIDI is not working. 
First of all, journalctl -f reports this when the esp32 is connected:
```
Dec 13 15:05:58 Ulmo kernel: usb 1-1: new full-speed USB device number 16 using xhci_hcd
Dec 13 15:05:58 Ulmo kernel: usb 1-1: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
Dec 13 15:05:58 Ulmo kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 13 15:05:58 Ulmo kernel: usb 1-1: Product: USB JTAG/serial debug unit
Dec 13 15:05:58 Ulmo kernel: usb 1-1: Manufacturer: Espressif
Dec 13 15:05:58 Ulmo kernel: usb 1-1: SerialNumber: EC:DA:3B:9D:10:08
Dec 13 15:05:58 Ulmo kernel: cdc_acm 1-1:1.0: ttyACM1: USB ACM device
Dec 13 15:05:58 Ulmo mtp-probe[433241]: checking bus 1, device 16: "/sys/devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1"
Dec 13 15:05:58 Ulmo mtp-probe[433241]: bus: 1, device: 16 was not an MTP device
Dec 13 15:05:58 Ulmo mtp-probe[433242]: checking bus 1, device 16: "/sys/devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1"
Dec 13 15:05:58 Ulmo mtp-probe[433242]: bus: 1, device: 16 was not an MTP device
```

It seems it is not recognized as a midi-capable device

Then I configured a simple action to send a program change on button press, and on the serial monitor and on the ESP32 display I can see that the action has been taken, but no event is received on the PC (double checked sniffing USB with wireshark and usbmon)

To further investigate, how am I supposed to flash the firmware file coming with the repo (firmware/lilygo-t-display-s3/firmware.bin.md5)?

Thank you in advance

alf45tar, 2023-12-13:
From `platformio.ini` remove 
```
-D USE_TINYUSB
-D CFG_TUD_ENABLED
-D CFG_TUD_MIDI
-D CFG_TUSB_CONFIG_FILE="${sysenv.HOME}/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/arduino_tinyusb/include/tusb_config.h"
```

and add
```
-D ARDUINO_USB_CDC_ON_BOOT=1
-D CORE_DEBUG_LEVEL=1
-D ARDUINO_USB_MODE=0
```

zavorra, 2023-12-14:
It is working great, thanks

closed 2023-12-18 by alf45tar