alf45tar/PedalinoMini #481: unable to end compile

davidin73, 2023-12-22:
Hello
I have a compile error that i can't solve.

**In file included from .pio/libdeps/esp32doit-devkit-v1/TFT_eSPI/TFT_eSPI.h:101:0,
                 from .pio/libdeps/esp32doit-devkit-v1/TFT_eSPI/TFT_eSPI.cpp:16:
.pio/libdeps/esp32doit-devkit-v1/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:14:25: fatal error: hal/gpio_ll.h: No such file or directory
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\libf68\TFT_eSPI\TFT_eSPI.cpp.o] Error 1
================================================= [FAILED] Took 203.59 seconds =================================================
Environment          Status    Duration
-------------------  --------  ------------
esp32doit-devkit-v1  FAILED    00:03:23.587
============================================ 1 failed, 0 succeeded in 00:03:23.587 ============================================ 

 *  The terminal process "C:\Users\Davide\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32doit-devkit-v1'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.**

alf45tar, 2023-12-22:
Add

`#include <driver/rtc_io.h>`

in `PedalinoMini.cpp` line 43 under

`#include <esp_bt_main.h>`

davidin73, 2023-12-23:
Thanks ALf45tar,buta same problem...

Compiling .pio\build\esp32doit-devkit-v1\lib1a1\MIDI Library\MIDI.cpp.o
Archiving .pio\build\esp32doit-devkit-v1\lib891\libResponsiveAnalogRead.a
Compiling .pio\build\esp32doit-devkit-v1\libf63\AppleMIDI\AppleMIDI.cpp.o
In file included from .pio/libdeps/esp32doit-devkit-v1/TFT_eSPI/TFT_eSPI.h:101:0,
                 from .pio/libdeps/esp32doit-devkit-v1/TFT_eSPI/TFT_eSPI.cpp:16:
.pio/libdeps/esp32doit-devkit-v1/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:14:25: fatal error: hal/gpio_ll.h: No such file or directory
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\libf68\TFT_eSPI\TFT_eSPI.cpp.o] Error 1
========================================================== [FAILED] Took 215.66 seconds ==========================================================
Environment          Status    Duration
-------------------  --------  ------------
esp32doit-devkit-v1  FAILED    00:03:35.657
===================================================== 1 failed, 0 succeeded in 00:03:35.657 ===================================================== 

 *  The terminal process "C:\Users\Davide\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32doit-devkit-v1'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

alf45tar, 2023-12-23:
Try version 3.2.0 just released

davidin73, 2023-12-24:
Hello!!!! 
Tried with 3.2.0
Works like a charm!!!!!
Now I must read the web UI instructions, you have added a lot of features from my last attempt, two years ago😅
Thank you very much @alf45tar , and merry Christmas to you and all your family!

closed 2023-12-24 by alf45tar