alf45tar/PedalinoMini #485: Using bank names when # of Pedals is under 6 causes boot loop/crash (bug)

Pedalnetics, 2023-12-27:
When setting:
`#define PEDALS                6`
to something less than 6 and then setting a bank name in Actions, the device will reboot and never load to the bank name screen. It loads the Wifi config bar to 100%, sits there, and eventually reboots itself.

alf45tar, 2023-12-27:
I need more information of your configuration to investigate the issue. I tried my T-Display with 5 pedals and it works.
Can you attach the log?

Pedalnetics, 2023-12-27:
Alrighty; I set my PEDALS define to 4, did a full clean, erase flash, build, upload, build filesystem image, and filesystem image upload.
On first boot, I SmartConfig the WiFi, everything boots normal/fine, and I'm able to get to the webpage. If I simply visit the Actions page and set a bank name and apply/save (in this case, my bank name was HX), it'll sit on the original A01 screen for a bit (not changing the bank name on the screen), and eventually reboot itself. Then it reconnects to wifi, sits there on the "Connecting to" screen with a full bar, and eventually reboots.
If I change the PEDALS define to anything under 6, this happens. if I set it to 6 or above, it works fine.
Logs from it booting and a crash:
```

SYSTEM_EVENT_SCAN_DONE
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
E (383) esp_core_dump_flash: No core dump partition found!
E (383) esp_core_dump_flash: No core dump partition found!
PedalinoMini™ - Version 3.1.13
CPU0 reset reason: POWERON_RESET Vbat power on reset
CPU1 reset reason: POWERON_RESET Vbat power on reset
PSRAM not found
ChipRevision 0, CPU Freq 240 MHz, SDK Version v4.4.3
Flash Size 8388608, Flash Speed 80000000 Hz
Internal Total Heap 299496, Internal Free Heap 269624
PSRAM Total Heap 0, PSRAM Free Heap 0
Default Vref: 1100mV
nvs Total 65536
PlatformIO Built Environment: heltec_wifi_kit_32_V3
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___ 
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \ 
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \ 
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  / 
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/ 
                                                                                  (c) 2018-2023 alf45star     
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: E4E22748
PSRAM not found
Internal Total Heap 299464, Internal Free Heap 269528
PSRAM Total Heap 0, PSRAM Free Heap 0
SPIFFS mount OK
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:      E4E22748
[NVS][Global][Boot Mode]:        1
[NVS][Global][BLE Server]:       
[NVS][Global][STA SSID]:         SN12IoT
[NVS][Global][STA Password]:     gDvR6v6XJLEU
[NVS][Global][AP SSID]:          Pedalino-E4E22748
[NVS][Global][AP Password]:      E4E22748
[NVS][Global][HTTP Username]:    admin
[NVS][Global][HTTP Password]:    pedalnetics
[NVS][Global][Bootstrap Theme]:  bootstrap
[NVS][Global][Current Profile]:  0
[NVS][Global][Flip Screen]:      0
[NVS][Global][Screen Timeout]:   3600000
[NVS][Global][Tap Dance Mode]:   0
[NVS][Global][Bank Switch]:      0
[NVS][Global][Debounce Time]:    5
[NVS][Global][SimultaneousGap]:  50
[NVS][Global][Single Time]:      200
[NVS][Global][Double Time]:      400
[NVS][Global][Long   Time]:      500
[NVS][Global][Repeat Time]:      1000
[NVS][Global][Encoder Sensit]:   5
[NVS][Global][RGB Order]:        10
[NVS][Global][LedsOnBright]:     5
[NVS][Global][LedsOffBright]:    1
[... 161 lines cut ...]
SSID        : SN12IoT
Password    : gDvR6v6XJLEU
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_STA_START
[  1737][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 202 - AUTH_FAIL
SYSTEM_EVENT_STA_DISCONNECTED
SYSTEM_EVENT_STA_CONNECTED
BSSID       : C6:A7:B9:14:D1:C4
RSSI        : -56 dBm
Channel     : 1
STA         : 48:27:E2:E4:83:C0
SYSTEM_EVENT_STA_GOT_IP
Hostname    : E4E22748
IP address  : 192.168.12.34
Subnet mask : 255.255.255.0
Gataway IP  : 192.168.12.1
DNS 1       : 192.168.12.251
DNS 2       : 0.0.0.0
mDNS responder started
BLE MIDI service advertising started (server mode)
Internal Total Heap 294032, Internal Free Heap 148644
[  9007][E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[  9012][E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[  9018][E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
```

Pedalnetics, 2023-12-27:
Does it have something to do with line 884 in the DisplayOLED.h file? It has a static-define of 6 pedals:
[image]

alf45tar, 2023-12-28:
You are right. I will change to

`const byte Pedals = min(PEDALS, 6);`

closed 2023-12-28 by alf45tar