alf45tar/PedalinoMini #488: Default "else" catchall config does not set FACTORY_DEFAULT_PIN mode for Long Press

Pedalnetics, 2024-01-03:
Firmware 3.2.0, Config.h, Line ~990 where it's setting up the default Actions for the onboard button for Profile+, Profile-, and Device Info, it's never setting the PEDALS-1 button to be PED_PRESS_1_2_L, therefor the LONG press to show device info is not working and you can't get to see the IP address of the device.

I fixed it in my local code by adding the following, not sure if it's the right fix or not but it did what I needed:
`  pedals[PEDALS-1].pressMode = PED_PRESS_1_2_L;`

alf45tar, 2024-01-03:
Not sure if I got it.
The long press works for me in my T-Display and T-Display-S3 board. Your suggested fix is exactly the 990 line.

Pedalnetics, 2024-01-03:
Sorry, mixed up my lines.
Correct, the ttgo stuff is getting caught in the If on line 962 then setting that default pedal mode on 990.
For other boards, they hit the else on 1056, and need that fix around 1084

Pedalnetics, 2024-01-05:
If there is a way I can privately fork the sponsors repo and submit pull requests to it, I'd be happy to submit some of these fixes/other features. Along with this I have the heltec v3 board working, some display pedal-tag enhancements related to the rectangle-block drawing, and have some other features I'll be coding like a sequence-reset command. but as it sits, I can't pull-request those back since I'm working off the newer sponsors code base