alf45tar/PedalinoMini #489: When using Sequences that send differing CC values, the Current Value "bars" on bank screen are always at full

Pedalnetics, 2024-01-05:
Easier to show a video: https://youtube.com/shorts/elrdfUI-Jfs

The switch tag is CC6-###, and on the "last press" screen it shows the proper value and value bar, but when it switches back to the bank screen, the value bar is always at 127

Pedalnetics, 2024-02-22:
@alf45tar any thoughts on this?

alf45tar, 2024-02-23:
Did you modified the firmware?

Pedalnetics, 2024-02-23:
It occured on a fresh flash from the public repo to heltec v2

alf45tar, 2024-02-23:
Are you sure? I never released a 4 pedals firmware.

Pedalnetics, 2024-02-23:
It occured on a fresh flash which is what led me to grab my modified one that was in an enclosure and easily presentable/recordable. I have a fresh public-based-git-pull flashed with it occurring in my hands right now, flashed as of 14minutes ago, with it happening with all 6 switches/bars present on the screen.
For more reproduction information:
Each switch is set to step-by-step a sequence. The sequence consists of 2 entries, both CC commands, one with a value of 0 and one with a value of 127.  The button-pressed screen jumps back and forth as it should according to the sequence, and the command send out via MIDI jumps back and forth as well. It is only the "all 6 switches on a single screen" bank screen that is incorrect.

alf45tar, 2024-02-24:
Fixed in version 3.2.5

closed 2024-02-24 by alf45tar