alf45tar/PedalinoMini #491: BUG Report: Incomplete Configuration Download

wcraffonara, 2024-01-21:
Hi,
  I'm using Firmware V3.1.13 and when I download my configuration using the <Download> button  in the <Configurations> page, it downloads only a part of the whole configuration! The downloaded config file ends with ",{"Sequence":13". 

Here is the content of the downloaded file (I've replace some Wifi values):

`{"Globals":[{"Hostname":"XXXX","BootMode":3,"BLEServer":"","Profile":0,"SSID":"XXXX","WiFiPassword":"xxxxx","SSIDsoftAP":"xxxxx","PasswordSoftAP":"xxxxxx","HTTPUsername":"xxxxx","HTTPPassword":"xxxxxx","Theme":"bootstrap","DebounceInterval":5,"SimultaneousGapTime":50,"PressTime":200,"DoublePressTime":400,"LongPressTime":500,"RepeatPressTime":1000,"EncoderSensitivity":5,"FlipScreen":true,"ScreenSaverTimeout":60,"TapDanceMode":false,"RepeatOnBankSwitch":false,"TapDanceBank":true,"Leds":10,"RGBOrder":"RGB","LedsOnBrightness":5,"LedsOffBrightness":1,"OSCLocalPort":8000,"OSCRemoteHost":"255.255.255.255","OSCRemotePort":9000}],"Ladder":[{"Step":1,"Level":352},{"Step":2,"Level":533},{"Step":3,"Level":640},{"Step":4,"Level":723},{"Step":5,"Level":806},{"Step":6,"Level":1023},{"Step":7,"Level":1023}],"Pedals":[{"Pedal":1,"Mode":"Momentary 1","InvertPolarity":false,"PressMode":"1","LatchEmulation":false,"AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false},{"Pedal":2,"Mode":"Jog Wheel","InvertPolarity":false,"PressMode":"1","LatchEmulation":false,"AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false},{"Pedal":3,"Mode":"Momentary 2","InvertPolarity":false,"PressMode":"1L","LatchEmulation":false,"AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false},{"Pedal":4,"Mode":"None","InvertPolarity":false,"PressMode":"None","LatchEmulation":false,"AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false},{"Pedal":5,"Mode":"None","InvertPolarity":false,"PressMode":"None","LatchEmulation":false,"AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false},{"Pedal":6,"Mode":"None","InvertPolarity":false,"PressMode":"None","LatchEmulation":false,"AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false},{"Pedal":7,"Mode":"None","InvertPolarity":false,"PressMode":"None","LatchEmulation":false,"AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false}],"Controls":[{"Control":1,"Pedal1":1,"Button1":1,"Pedal2":0,"Button2":0,"Led":0},{"Control":2,"Pedal1":2,"Button1":1,"Pedal2":0,"Button2":0,"Led":0},{"Control":3,"Pedal1":3,"Button1":1,"Pedal2":0,"Button2":0,"Led":0},{"Control":4,"Pedal1":3,"Button1":2,"Pedal2":0,"Button2":0,"Led":0},{"Control":10,"Pedal1":0,"Button1":1,"Pedal2":0,"Button2":0,"Led":0},{"Control":37,"Pedal1":7,"Button1":1,"Pedal2":0,"Button2":0,"Led":0},{"Control":38,"Pedal1":7,"Button1":1,"Pedal2":0,"Button2":0,"Led":0},{"Control":39,"Pedal1":7,"Button1":1,"Pedal2":0,"Button2":0,"Led":0}],"BankNames":[{"Bank":0,"Name":"Global"},{"Bank":1,"Name":":1 > 1 1 1"},{"Bank":2,"Name":":2 > 0 0 0"},{"Bank":3,"Name":":3 > 1 0 0"},{"Bank":4,"Name":":4 > 0 1 0"},{"Bank":5,"Name":":5 > 0 0 1"},{"Bank":6,"Name":":6 > 1 1 0"},{"Bank":7,"Name":":7 > 0 1 1"},{"Bank":8,"Name":":8 > 1 0 1"},{"Bank":9,"Name":""},{"Bank":10,"Name":""},{"Bank":11,"Name":""},{"Bank":12,"Name":""},{"Bank":13,"Name":""},{"Bank":14,"Name":""},{"Bank":15,"Name":""},{"Bank":16,"Name":""},{"Bank":17,"Name":""},{"Bank":18,"Name":""},{"Bank":19,"Name":""},{"Bank":20,"Name":""}],"Actions":[{"Bank":0,"Control":3,"Led":255,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":0,"Event":"Press","Message":"Bank+","Channel":0,"Code":0,"Value1":1,"Value2":8,"OSCAddress":""},{"Bank":0,"Control":3,"Led":255,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":0,"Event":"Long Press","Message":"Profile+","Channel":0,"Code":0,"Value1":0,"Value2":2,"OSCAddress":""},{"Bank":0,"Control":4,"Led":255,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":0,"Event":"Press","Message":"Bank-","Channel":0,"Code":0,"Value1":1,"Value2":8,"OSCAddress":""},{"Bank":0,"Control":4,"Led":255,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":0,"Event":"Long Press","Message":"Profile-","Channel":0,"Code":0,"Value1":0,"Value2":2,"OSCAddress":""},{"Bank":1,"Control":1,"Led":0,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":1,"Event":"Press","Message":"Sequence","Channel":1,"Code":0,"Value1":0,"Value2":0,"OSCAddress":""},{"Bank":2,"Control":1,"Led":0,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":6,"Event":"Press","Message":"Sequence","Channel":2,"Code":0,"Value1":0,"Value2":0,"OSCAddress":""},{"Bank":3,"Control":1,"Led":0,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":3,"Event":"Press","Message":"Sequence","Channel":3,"Code":0,"Value1":0,"Value2":0,"OSCAddress":""},{"Bank":11,"Control":1,"Led":255,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":6,"Event":"None","Message":"None","Channel":1,"Code":0,"Value1":0,"Value2":127,"OSCAddress":""},{"Bank":12,"Control":1,"Led":255,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":6,"Event":"None","Message":"None","Channel":1,"Code":0,"Value1":0,"Value2":127,"OSCAddress":""},{"Bank":13,"Control":1,"Led":255,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":6,"Event":"None","Message":"None","Channel":1,"Code":0,"Value1":0,"Value2":127,"OSCAddress":""},{"Bank":14,"Control":1,"Led":255,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":6,"Event":"None","Message":"None","Channel":1,"Code":0,"Value1":0,"Value2":127,"OSCAddress":""}],"Interfaces":[{"Interface":1,"Name":"USB MIDI   ","In":false,"Out":false,"Thru":false,"Clock":false,"ShowIncoming":false,"ShowOutcoming":false},{"Interface":2,"Name":"Legacy MIDI","In":true,"Out":true,"Thru":true,"Clock":false,"ShowIncoming":false,"ShowOutcoming":false},{"Interface":3,"Name":"RTP-MIDI   ","In":false,"Out":false,"Thru":false,"Clock":false,"ShowIncoming":false,"ShowOutcoming":false},{"Interface":4,"Name":"ipMIDI     ","In":false,"Out":false,"Thru":false,"Clock":false,"ShowIncoming":false,"ShowOutcoming":false},{"Interface":5,"Name":"BLE MIDI   ","In":false,"Out":false,"Thru":false,"Clock":false,"ShowIncoming":false,"ShowOutcoming":false},{"Interface":6,"Name":"OSC        ","In":false,"Out":false,"Thru":false,"Clock":false,"ShowIncoming":false,"ShowOutcoming":false}],"Sequences":[{"Sequence":1,"Step":1,"Message":"Control Change","Channel":1,"Code":102,"Value":127,"Led":0,"Color":"#000000"},{"Sequence":1,"Step":2,"Message":"Control Change","Channel":1,"Code":103,"Value":127,"Led":0,"Color":"#000000"},{"Sequence":1,"Step":3,"Message":"Control Change","Channel":1,"Code":104,"Value":127,"Led":0,"Color":"#000000"},{"Sequence":1,"Step":4,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":1,"Step":5,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":1,"Step":6,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":1,"Step":7,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":1,"Step":8,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":1,"Step":9,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":1,"Step":10,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":2,"Step":1,"Message":"Control Change","Channel":1,"Code":102,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":2,"Step":2,"Message":"Control Change","Channel":1,"Code":103,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":2,"Step":3,"Message":"Control Change","Channel":1,"Code":104,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":2,"Step":4,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":2,"Step":5,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":2,"Step":6,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":2,"Step":7,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":2,"Step":8,"Message":"None","Channel":0,"Code":0,"Value":0,"Led":0,"Color":"#000000"},{"Sequence":2,"Step":9,"Message":"None","Channel":0," [... cut ...] el":1,"Code":0,"Value":0,"Led":1,"Color":"#000000"},{"Sequence":13,"Step":2,"Message":"None","Channel":1,"Code":0,"Value":0,"Led":1,"Color":"#000000"},{"Sequence":13,"Step":3,"Message":"None","Channel":1,"Code":0,"Value":0,"Led":1,"Color":"#000000"},{"Sequence":13,"Step":4,"Message":"None","Channel":1,"Code":0,"Value":0,"Led":1,"Color":"#000000"},{"Sequence":13,"Step":5,"Message":"None","Channel":1,"Code":0,"Value":0,"Led":1,"Color":"#000000"},{"Sequence":13`

How can I fix this?

Thanks 
Walter

alf45tar, 2024-01-22:
It is usually a memory issue. Which is the % of SPIFFS available?
Did you check the log for error when saving the configuration?

wcraffonara, 2024-01-22:
Currently used is 54% of SPIFFS. This should be low enough, or?
You mean the error log from debug interface? No I've not. I will connect it to my PC and will take a look to the debug interface.

wcraffonara, 2024-01-22:
I've connected the PedalinoMini to my PC and when I navigate to the <Configurations> page then it prints following on the terminal (6x Looking for configuration files...):
[image]

In the Configuration Editor the last Sequence value seems to be incomplete:
[image]

Can you explain how to check the "log for error"?

wcraffonara, 2024-01-22:
I've removed/deleted the corrupted (?) Sequence number 125 directly in the Configurations Editor and then I can export a valid Configuration file.
I've no idea why the Sequence 125 was corrupted.

alf45tar, 2024-01-23:
The configuration file is a JSON document allocate into memory before saving. Check the free memory in Home page and compare it with the file dimension. The file is trrncated because is bigger than the max block of free memory.

wcraffonara, 2024-01-23:
Since I've removed Sequence number 125 I can download a valid JSON file with a file size of 19kB. If I compare this file size with the reported free space from Home page then it looks like there is more than enough free memory:

[image]

alf45tar, 2024-02-04:
Did you tried the latest version?

wcraffonara, 2024-02-07:
Hi,
  yes, I have. The issue seems to be solved. I can download the entire configuration without any problems.

closed 2024-02-07 by wcraffonara