alf45tar/PedalinoMini #492: Unable to update firmware using web installer

mitya-murakhouski, 2024-01-30:
Hi, 

I'm trying to update firmware from v2.3.2 to v3.2.1 using web installer. After connecting Pedalino via USB I see in chrome://device-log/ the following:

[21:53:08] USB device added: path=/dev/bus/usb/001/020 vendor=4292 "Silicon Labs", product=60000 "CP2104 USB to UART Bridge Controller", serial="023752DF", guid=be084a61-941d-41ae-926d-a0b6f56286c6

But the installer say that 'No compatible devices found'. What could be the reason?

Thanks!

alf45tar, 2024-01-30:
Did you installed the driver https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads?

mitya-murakhouski, 2024-01-30:
Yes, drivers are included in linux kernel since 2.6
6.5.0-15-generic/kernel/drivers/usb/serial/cp210x.ko

lsusb | grep CP
Bus 001 Device 022: ID 10c4:ea60 Silicon Labs CP210x UART Bridge

BTW Previously I managed to install  v2.3.2 on my ttgo using web installer

mitya-murakhouski, 2024-01-31:
I was able to update from windows machine, so not sure that this is an issue.
Thanks for support and great Pedalino!

closed 2024-02-04 by alf45tar