alf45tar/PedalinoMini #498: 3.2.3 Firmware issues

bobvc133, 2024-02-19:
1.After upgrading firmware from 3.1.13 to 3.2.3, the problem of restarting or invalid application saving often occurs when applying & saving configurations. This problem exists on both TTGO T DISPLAY and LILYGO T DISPLAY S3 development boards. The logs and files are as follows

[image]
[Default.TXT](https://github.com/alf45tar/PedalinoMini/files/14326127/Default.TXT)

2.How to use OSC Message event in action?

[image]

3.When configuring the action of sending OSC Message, the Tag configuration is invalid and nothing is displayed on the operation button screen.

alf45tar, 2024-02-23:
I uploaded your Default.TXT file into my T-Display S3 without reset.

bobvc133, 2024-02-25:
> I uploaded your Default.TXT file into my T-Display S3 without reset.

I tested three boards, two TTGO T DISPLAY and one LILYGO T DISPLAY S3 development board. The 3.2.2 firmware also has this problem, and the probability of restarting is 90%.

alf45tar, 2024-02-25:
Your file requires 41942 bytes to be uploaded. it can be applied the first time but not a second time without a reboot. I think due to memory fragmentation. I suggest to create 2 files: one with the 160 sequences and one with the rest to keep each of them enough small.

alf45tar, 2024-02-26:
Check version 3.2.6. It should be fixed.

bobvc133, 2024-02-27:
> Check version 3.2.6. It should be fixed.

After testing the 3.2.7 firmware, the TTGO T DISPLAY problem still exists. Clicking Apply to save the configuration will cause a restart, or it will prompt success, but the action page will have no content. LILYGO T DISPLAY S3 All three ABC user configuration groups can successfully apply the saved configuration, but switching the ABC user configuration will cause a restart or repeated restarts.

alf45tar, 2024-02-27:
If the Actions page has no content check log for 

```
Reading` /Default.TXT.cfg from SPIFFS ... deserializeJson() failed with code NoMemory

bobvc133, 2024-02-28:
[image]

bobvc133, 2024-02-28:
> If the Actions page has no content check log for
> 
> ```
> Reading` /Default.TXT.cfg from SPIFFS ... deserializeJson() failed with code NoMemory
> ```

[image]

alf45tar, 2024-03-03:
Reboot has been solved in 3.2.8.
NoMemory issue can be solved uploading small files and use append to current configuration.

bobvc133, 2024-03-03:
> Reboot has been solved in 3.2.8. NoMemory issue can be solved uploading small files and use append to current configuration.

After repeated testing of the 3.2.8 firmware, my "Default" configuration file still has problems with the operation page being empty or restarting when applied & saved on the TTGO T DISPLAY development board. The same is true for uploading a small file and attaching it to the current configuration. But if you use firmware 3.1.13, this problem will basically not occur, and the success rate is more than 90%.
  LILYGO T DISPLAY S3 development board issue has been fixed.

bobvc133, 2024-03-12:
After testing 3.2.9 firmware problem still exists

bobvc133, 2024-04-05:
After testing 3.3.1 firmware problem still exists

alf45tar, 2024-04-06:
I am not providing support for your commercial projects here.

closed 2024-04-06 by alf45tar

bobvc133, 2024-04-06:
> I am not providing support for your commercial projects here.

PedalinoMini is indeed a great project. I admit that I did make some for sale on second-hand trading platforms, but I did not regard PedalinoMini as a commercial project. I did not change the project name. The external introduction will also indicate that the project comes from GITHUB, and I will also give Other DIY users provide help, because I am a perfectionist, have high requirements for DIY, and do not like to piece together randomly, so I can only find a factory to customize a batch of shells and accessories with a minimum order quantity, but I only need one, and the rest I have to find a way to sell it. I just want to maintain a balance that can meet the ever-changing DIY requirements while recovering the extra investment. To be honest, the PedalinoMini project is indeed great, but it is difficult to become a commercial project. First of all, it is not yet mature. Secondly, the market demand is too small. There are already many mature similar products. PedalinoMini can only exist among DIY people, otherwise You may be able to mass-produce and sell products immediately, instead of obtaining funds for continued research and development through project sponsorship. So far, I have not made any money from the PedalinoMini project. On the contrary, I have invested a lot of time and money to let more people know about the PedalinoMini project. I am just doing this alone. In addition to testing program bugs, I also need to design circuits, weld tests, produce tutorials, online guidance and teaching, and deal with user problems. If you think this is a so-called commercial project, I can stop immediately. From the time I found this project to now, I have spent thousands of dollars on DIY. I am tired of the endless use of teaching' after-sales and lack of funds. At first, I just wanted to DIY a powerful MIDI pedal for myself, but I didn’t expect that I got deeper and deeper into it.