alf45tar/PedalinoMini #500: Upload fail "A fatal error occurred: This chip is ESP32-S3(beta3) not ESP32. Wrong --chip argument?"

upgrader-dev, 2024-02-23:
Hello .
First things first, thank you for your work. I hope I will be able to enjoy it soon :)
I have a Adafruit Feather ESP32-S3 Reverse TFT.
This board is supported by [Arduino core for ESP32](https://github.com/espressif/arduino-esp32).

I followed your guide to build & upload.

Just before upload I connect the board via USB. And Then I get this error : 

"A fatal error occurred: This chip is ESP32-S3(beta3) not ESP32. Wrong --chip argument?".

Don't know why the upload hook thinks it is ESP32 and not ESP32-S3.

Any idea on how to solve this?

alf45tar, 2024-02-23:
Your board is not supported by PedalinoMini.
The only S3 supported board are BPI Leaf S3 or T-Display S3. Which firmware did you used?

For S3 boards remember to enter Bootloader mode to enable flash via USB-CDC.

1. connect it via USB,
2. hold BOOT button,
3. press RST button once,
4. then release BOOT button.
5. Upload firmware
6. RST to exit Bootloader mode

upgrader-dev, 2024-02-23:
I used esp32doit-devkit-v1. 
Maybe you could modify the README.md as it stands that  : 
"Any boar supported by [Arduino core for ESP32](https://github.com/espressif/arduino-esp32) is compatible for PedalinoMini"

And my board is in the supported board above.

And yes I was entering the bootloader mode.

I'll go for another board, thank you for your response.

closed 2024-02-23 by upgrader-dev