alf45tar/PedalinoMini #501: Display get stuck at the loading bar

ukudos, 2024-03-04:
Hello alf45tar,
After my Pedalino worked well for several years, I now wanted to install the latest version (3.2.2). Using PC USB and repository everything worked without any error messages.
Unfortunately, the display (sh1106) gets stuck at the loading bar when selecting the boot menus.
For example, if I choose AP mode, I can select the AP on the iPad and edit and save it in the web interface. Only the display on the pedalino shows nothing or gets stuck at the loading bar.
Any idea what's going wrong here?
Looking forward to an answer, greetings!
[FEBF713C_log.txt](https://github.com/alf45tar/PedalinoMini/files/14481925/FEBF713C_log.txt)

alf45tar, 2024-03-06:
I do not see any reason in the log.
Which board are you using? Did you modified the firmware?

ukudos, 2024-03-06:
Many thanks for your response.
Doit ESP32 DEVKIT V1 is the board.
In platformio i set the booflags for the ssh1106 display.
Greetings

alf45tar, 2024-03-11:
I do not own anymore a ESP32 board (with the exception of T-Display) where I can test it.

ukudos, 2024-03-15:
Is there a recommendation to what the latest firmware version is for an ESP32 board?

alf45tar, 2024-03-15:
The recommend firmware is always the latest version because I usually support only it.
I cannot check every board on new release. I usually test it on T-Display and T-Display-S3. For new build I suggest LilyGo T-Display-S3 only.

raidolo, 2024-05-05:
Same here... same display same issue. 

ESP-WROOM-32
SH1106 1.3" display
Env: Doit ESP32 DEVKIT V1

alf45tar, 2024-05-06:
The only board with SH1106 I still own is TTGO T-Eight. I tested with the latest version 3.3.2 (just released) and it works.

raidolo, 2024-05-06:
@alf45tar foeget it, I just saw the invitation... thanks

raidolo, 2024-05-06:
I've done some tests... if you flip the screen then flip it back it works while powered on, but then at the next reboot it stucks again at the wifi connection bar.

alf45tar, 2024-05-07:
> @alf45tar I got the invite and I was able to access the sponsors repo. Could you please add me to the Pedalino-8 STLs repo please?

For PedalinoMini 4/6/8 I have a different supporting model than for PedalinoMini Case 1.
PedalinoMini Case 1 is provided to all sponsors but PedalinoMini 4/6/8 requires a separate supporting fee.

- 9 USD for PedalinoMini 4
- 12 USD for PedalinoMini 6
- 14 USD for PedalinoMini 8

 I received your 3 years sponsorship. Do you want me to allocate it to PedaliniMini 8 + 1 year of sponsorship?

raidolo, 2024-05-07:
Hi @alf45tar, I've just built the latest 3.3.2 version and the display issue is fixed on my SH1106. Now I can see also the logos of WiFI and BT on boot.

closed 2024-05-16 by alf45tar