alf45tar/PedalinoMini #502: first led doesn't work since 2.3.2 Liligo T Display S3

DxC82, 2024-03-10:
I built the Pedalinomini 4.  unfortunately the 7 LEDs are not assigned correctly.  If the first one should light up, the 2nd one lights up, the third one instead of the 2nd and so on.

alf45tar, 2024-03-10:
Are you saying that led number 1 is the second, led 2 is the third, etc. and you cannot light up first led?

DxC82, 2024-03-10:
Yeah right.  The number of LEDs cannot be changed via the html configuration.  The number 10 always remains.

DxC82, 2024-03-10:
Already after flashing, in the first boot phase, the first LED is omitted.  in version 2.3.1 it still lights up.  from version 2.3.2 no longer.

DxC82, 2024-03-10:
I flashed version 2.3.1 again.  4 controls, 4 actions with PC, led on and led off. led 7 for button 1. led 5 for button 2, led 3 for button 3 and led 1 for button 1. the LEDs come on correctly but don't go off when switching.
attached the config file. 
[config.zip](https://github.com/alf45tar/PedalinoMini/files/14551461/config.zip)

alf45tar, 2024-03-10:
Leds are numbered from right to left to simplify the internal connections.
I confirm there is an issue with PC and leds.

alf45tar, 2024-03-11:
As a temporary fix add your PC actions on a different bank. They are not light off led when used in bank 0.

alf45tar, 2024-03-11:
Fix in 3.2.9 just released

alf45tar, 2024-03-13:
Are you using a WS2812B led strip?

DxC82, 2024-03-13:
Yes.
WS2812B LED Strip, 
Model: BTF-5V-030L-B 
FNSKU: X000U6VGJV 
SKU: WS2812B5M30LB30
5 Meter Roll.

DxC82, 2024-03-14:
The problem only exists with my Lilygo T-Display S3 (PedalinoMini 4 project).  My Lilygo T display works.

alf45tar, 2024-03-14:
Check the connections. Can you post some shots?

DxC82, 2024-03-14:
[image]

alf45tar, 2024-03-14:
Connections look good. I do not think it is a board issue.

alf45tar, 2024-03-15:
Did you try to upload any example from FastLed library?

DxC82, 2024-03-15:
You're right, it's not the board.  new board and the same error.  There must be something wrong with the board-specific configuration, right?
[image]
The first LED is also left out there.

DxC82, 2024-03-15:
I haven't tried any other code yet.  I'll test FastLED examples right away.

DxC82, 2024-03-17:
The Adafruit NeoPixel  buttoncycler works! 
[image]

[image]

https://github.com/alf45tar/PedalinoMini/assets/113459859/a24ebce9-d16f-4183-a708-40377242e681

DxC82, 2024-03-17:
https://github.com/alf45tar/PedalinoMini/assets/113459859/287b1d7d-d7fa-4297-96df-46de6c03b5ba

DxC82, 2024-03-17:
sorry for closing and opening. the LEDs were too bright, I hit the wrong button.

DxC82, 2024-03-17:
The FastLED example ColorPalette works too! 

[image]

https://github.com/alf45tar/PedalinoMini/assets/113459859/20c0a0c6-be2c-4ad2-8ce3-98783ed3d795

DxC82, 2024-03-17:
Works with WS2812 and WS2812B , 7 or 10 Num_Leds.. 
[image]

alf45tar, 2024-03-23:
Can you try the blueboard4 configuration from latest firmware?

DxC82, 2024-03-23:
Here with the config blueboard4. 

https://github.com/alf45tar/PedalinoMini/assets/113459859/271e385c-342c-4dea-85f7-beec12fb36a2

alf45tar, 2024-03-30:
Doest it work in 3.2.1?

DxC82, 2024-03-30:
Yes on 3.2.1 first led works. Since 3.2.2 doesn't, like the videos..

alf45tar, 2024-03-31:
I retested again my PedalinoMini 4 with 3.2.1 and 3.2.2 and latest firmware and there is no difference for led 1.
Very hard to debug this issue.

DxC82, 2024-04-02:
I retested again, too.. 
3.2.1 led order OK. 
[image]

3.2.2 led order wrong. First led without function. 
[image]

[image]

I can't believe I'm the only one with this problem..

alf45tar, 2024-04-02:
I found the potential reason. Can you try to disable Legacy MIDI OUT?

DxC82, 2024-04-02:
i disabled first the Lagacy Midi Out -- no change.
i disabled all Midi Interfaces exept BLE in and out -- no change

When booting, the first LED is already deactivated. Maybe this information will help

DxC82, 2024-04-02:
I am currently on version 3.2.2

DxC82, 2024-04-02:
I have now compiled version 3.3.0 with midi out on pin 21 (line 117 Pedalino.h)
and it works! I don't know if pin21 is a good choice, but here lies the mistake!

Only the html configurations page now looks strange in my test build..

DxC82, 2024-04-04:
Thanks, the issue is fixed with 3.3.1

closed 2024-04-04 by DxC82