alf45tar/PedalinoMini #504: Please provide wiring diagram for PedalinoMini4

buzzyc, 2024-03-15:
I am excited to build this device, however other than the 3D render, I cannot see any other wiring diagram.
This wouldn't be so bad if the render didn't obscure some of the wiring locations on the S3 shield.
Could you please provide a flat image of the wiring pinout for the S3 please?

Cheers :)

mknerr, 2024-03-15:
The diagrams in the ReadMe don't suffice?

buzzyc, 2024-03-15:
As per this image: 
https://github.com/alf45tar/PedalinoMini-4/blob/main/images/PedalinoMini%204%20Wiring.jpg

The top row is obscured. I could probably extrapolate, but it would be nice to not have to guess pinouts

alf45tar, 2024-03-15:
[image]

alf45tar, 2024-03-15:
I updated the wiring image into the repository too.

closed 2024-03-15 by alf45tar