alf45tar/PedalinoMini #505: WEB interface flashing firmware problem

jsonslim, 2024-04-03:
Hello,

The problem: after using http://alf45tar.github.io/PedalinoMini/installer     and connecting to my router, the page in my case - http://116de378.local/    doesn't load, the browser shows this: 
[image]

I can see the board in my home network with 192.168.0.46, and when i open the page, it shows this:  
[image]

Desired behavior:   open the page from the board, without nessecity of entering login and password, because I have no idea where to get it.

Thanks!

alf45tar, 2024-04-03:
Username is admin
Default password is 116DE378

Thanks for the suggestion to disable authentication on first login.

jsonslim, 2024-04-03:
Thank you for the answer! 
After I wrote this issue I saw that the admin/pass instructions were written on the readme section. But the  http://116de378.local/    still doesn't load, it's a timeout problem...

alf45tar, 2024-04-03:
I suggest to use some tool to debug mDNS/Bonjour/Zeroconf.
I am using https://itunes.apple.com/us/app/discovery-dns-sd-browser/id1381004916?mt=12 on macOS and 
https://www.tobias-erichsen.de/software/zeroconfservicebrowser.html on Windows.

closed 2024-05-16 by alf45tar

serhatsoyyigit, 2024-06-14:
This is very confusing. It makes no sense to have a mandatory password for the first installation. I've been trying to figure this out for hours and it doesn't say anywhere in the description. On the contrary, it says that no information is required for login.

How can you try so hard to make it simple and make such a mistake in this part?

alf45tar, 2024-06-14:
Default password has been removed

jsonslim, 2024-06-14:
That's great! Thank you!