alf45tar/PedalinoMini #507: PIN 12 HIGH on boot

upgrader-dev, 2024-04-10:
Hello.

I'm bulding a simple Pedalino with 4 buttons.

The pedal 4 is linked to the PIN 12 on TTGO T-display.

But when I connect this PIN 12 to my footswitch, the boot is impossible.

If I switch to pedal 5, everything works flawlessly.

Lately I found this issue with LilYGO that says that PIN 12 HIGH is used to determine if boot lust be done from SD card.
https://github.com/LilyGO/TTGO-T5-ePaper1.54/issues/3

What do you guys think about that?

upgrader-dev, 2024-04-12:
[image]

alf45tar, 2024-04-14:
The permanent solution is here even if I am not using it on my board. 

https://github.com/espressif/esp-idf/tree/release/v3.2/examples/storage/sd_card#note-about-gpio12

upgrader-dev, 2024-04-14:
Thank you. So, just to check I understood... I have to insert this line 
[    components/esptool_py/esptool/espefuse.py set_flash_voltage 3.3V]() in the right file to force the 3.3V function. 

Which file do I have to insert this line in?

alf45tar, 2024-04-14:
I think it is an one time only command. Setting is permanent stored into efuse and cannot changed anymore.

closed 2024-05-16 by alf45tar