alf45tar/PedalinoMini #510: Version 3.3.1 - 3.3.2 - Action Page strange behaviour - ESP32-DEVKIT

raidolo, 2024-05-06:
Hi @alf45tar, 

I'm having a very difficult time using firmware 3.3.1 or 3.3.2. I've erased the flash of my ESP32, I've built the firmware, uploaded the filesystem image, except the display issue I have with the SH1106 display, discussed in the other "issue" thread, it seems working fine. But, as soon I go to the actions page, I'm starting to have weird issues. Most of the times I need to reload the page multiple times, I've delete the default actions on the Control 38 and as soon as I've clicked on the trash icon, I got a blank page, then the CSS didn't load and I got something like this: 

[image]

Every time weird stuff happen, I also had a backtrace at some point while deleting the default action. This is the output from the console:

```

Looking for configuration files on SPIFFS root ...
/case1.cfg
/mustang.cfg
/blueboard.cfg
/blueboard4.cfg
done.
[W][AsyncTCP.cpp:930] _poll(): pcb is NULL
abort() was called at PC 0x401e34f7 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40090854:0x3ffe9ae0 0x40090ad1:0x3ffe9b00 0x401e34f7:0x3ffe9b20 0x401e353e:0x3ffe9b40 0x401e2c17:0x3ffe9b60 0x401e2f4e:0x3ffe9b80 0x401e2cb5:0x3ffe9ba0 0x4010b56a:0x3ffe9bc0 0x40106e57:0x3ffe9c00 0x40107a1c:0x3ffe9c40 0x40107a57:0x3ffe9c60 0x401085a2:0x3ffe9c80 0x40091b12:0x3ffe9cd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
PedalinoMini™ - Version 3.3.1
CPU0 reset reason: SW_CPU_RESET Software reset CPU
CPU1 reset reason: SW_CPU_RESET Software reset CPU
PSRAM not found
```

Is this something expected? 
I did multiple fresh installs erasing or doing the factory reset, but the behaviour is the same with 3.3.1 or 3.3.2.

Thanks a lot in advance.

Regards,
Alessio.

raidolo, 2024-05-07:
> Hi, i had the same problem. Try to boot on "WiFi only", this action resolved my connection.

Do you believe it's a latency/connection issue? It's something I've noticed as well, I was pinging the ip and I got high ping times, while pretty close to the access point. I was thinking about a high cpu usage. I'll try the wifi only mode.

alf45tar, 2024-05-07:
Can you give a try to the latest version?

raidolo, 2024-05-07:
It seems ok with the latest version and WIFI only mode. The ping to the board is now normal (wifi only) and I was able to delete three banks of actions via the webUI without any weird behavior, and a pretty snappy interface I must say. I still have issues while booting the board with normal mode. The ping goes up like crazy and the action page is unusable:

Wifi only mode:

```
PING 192.168.1.217 (192.168.1.217) 56(84) bytes of data.
64 bytes from 192.168.1.217: icmp_seq=2 ttl=255 time=307 ms
64 bytes from 192.168.1.217: icmp_seq=3 ttl=255 time=2.66 ms
64 bytes from 192.168.1.217: icmp_seq=4 ttl=255 time=2.77 ms
64 bytes from 192.168.1.217: icmp_seq=5 ttl=255 time=2.73 ms
64 bytes from 192.168.1.217: icmp_seq=6 ttl=255 time=2.63 ms
64 bytes from 192.168.1.217: icmp_seq=7 ttl=255 time=2.66 ms
64 bytes from 192.168.1.217: icmp_seq=8 ttl=255 time=4.44 ms
64 bytes from 192.168.1.217: icmp_seq=9 ttl=255 time=7.13 ms
64 bytes from 192.168.1.217: icmp_seq=10 ttl=255 time=2.63 ms
64 bytes from 192.168.1.217: icmp_seq=11 ttl=255 time=2.60 ms
64 bytes from 192.168.1.217: icmp_seq=12 ttl=255 time=2.58 ms
64 bytes from 192.168.1.217: icmp_seq=13 ttl=255 time=2.75 ms
64 bytes from 192.168.1.217: icmp_seq=14 ttl=255 time=2.76 ms
^C
--- 192.168.1.217 ping statistics ---
14 packets transmitted, 13 received, 7.14286% packet loss, time 13014ms
rtt min/avg/max/mdev = 2.576/26.594/307.381/81.065 ms
```

Normal mode:

```
PING 192.168.1.217 (192.168.1.217) 56(84) bytes of data.
64 bytes from 192.168.1.217: icmp_seq=1 ttl=255 time=70.8 ms
64 bytes from 192.168.1.217: icmp_seq=2 ttl=255 time=94.2 ms
64 bytes from 192.168.1.217: icmp_seq=3 ttl=255 time=15.0 ms
64 bytes from 192.168.1.217: icmp_seq=4 ttl=255 time=38.2 ms
64 bytes from 192.168.1.217: icmp_seq=5 ttl=255 time=61.1 ms
64 bytes from 192.168.1.217: icmp_seq=6 ttl=255 time=84.2 ms
64 bytes from 192.168.1.217: icmp_seq=7 ttl=255 time=107 ms
64 bytes from 192.168.1.217: icmp_seq=8 ttl=255 time=27.8 ms
64 bytes from 192.168.1.217: icmp_seq=9 ttl=255 time=53.3 ms
64 bytes from 192.168.1.217: icmp_seq=10 ttl=255 time=74.1 ms
64 bytes from 192.168.1.217: icmp_seq=11 ttl=255 time=97.1 ms
^C
--- 192.168.1.217 ping statistics ---
11 packets transmitted, 11 received, 0% packet loss, time 10009ms
rtt min/avg/max/mdev = 15.031/65.736/107.168/28.429 ms
```
Maybe it's just too much work to open the web ui, listen for the BT and then open the webui, it's not a matter of network connectivity by itself but also of the CPU load like I suspected in the beginning. I dunno, if someone knows the reason is fine, but I'm more than happy to know how to apply the workarond, I'm not opening the web gui too much after I did the setup anyway...

alf45tar, 2024-05-14:
The latest version 3.3.3 is stable again with both WiFi and Bluetooth activated.

christianbratschke13, 2024-05-16:
Hey @alf45tar, could you tell me where I can find version 3.3.3? Somehow for me version 3.2.2 is the latest release. Thanks!

alf45tar, 2024-05-16:
The latest version 3.3.4 is available to sponsor only.
3.2.2 is the latest public version.

More details here https://github.com/alf45tar/PedalinoMini?tab=readme-ov-file#sponsors

closed 2024-05-16 by alf45tar