alf45tar/PedalinoMini #511: Version 3.3.2 - Configurations import - Sequence number increased by 1 or 2 in the imported action.

raidolo, 2024-05-07:
Hi @alf45tar, 

I just noticed another thing, happening in 3.3.2, but also in 3.2.2. 
I've saved the configurations of my actions to have a backup and while testing the display issue I had with the SH1106, I've imported the backups few times. Also because of the fact that I had the action page issue, at first I was thinking that something was wrong during the restore of the backups, but now that I have a working action page, I'm sure it could be a little bug in the actions restore, while in the action there is a sequence.

I have this saved in my conf file, basically 4 actions, 2 per each control, the first one sends the midi command, the second one sets the LEDs. 
```
   "Actions": [
    {
      "Bank": 1,
      "Control": 2,
      "Led": 255,
      "Color0": "#000000",
      "Color1": "#000000",
      "NameOff": "",
      "NameOn": "",
      "Slot": 0,
      "Event": "Press & Release",
      "Message": "Sequence",
      "Channel": 7,
      "Code": 0,
      "Value1": 0,
      "Value2": 0,
      "OSCAddress": ""
    },
    {
      "Bank": 1,
      "Control": 2,
      "Led": 0,
      "Color0": "#000000",
      "Color1": "#000000",
      "NameOff": "S1",
      "NameOn": "S1",
      "Slot": 6,
      "Event": "Press & Release",
      "Message": "Program Change",
      "Channel": 2,
      "Code": 0,
      "Value1": 0,
      "Value2": 0,
      "OSCAddress": ""
    },
    {
      "Bank": 1,
      "Control": 3,
      "Led": 255,
      "Color0": "#000000",
      "Color1": "#000000",
      "NameOff": "",
      "NameOn": "",
      "Slot": 6,
      "Event": "Press & Release",
      "Message": "Sequence",
      "Channel": 8,
      "Code": 0,
      "Value1": 0,
      "Value2": 0,
      "OSCAddress": ""
    },
    {
      "Bank": 1,
      "Control": 3,
      "Led": 0,
      "Color0": "#000000",
      "Color1": "#000000",
      "NameOff": "S2",
      "NameOn": "S2",
      "Slot": 6,
      "Event": "Press & Release",
      "Message": "Program Change",
      "Channel": 2,
      "Code": 0,
      "Value1": 1,
      "Value2": 1,
      "OSCAddress": ""
    }
  ]
```

But I got this in the Actions page, basically all the sequences got increased by 1: 

[image]

raidolo, 2024-05-15:
Hi @alf45tar, I've tested the 3.3.3, now the sequence in the actions are ok, but something else happened, now the sequences called by an action are increased by 1. For instance: I set sequence 3, but in reality it plays sequence 4. I'm testing a bit more, not importing the configurations and see what's happening...

raidolo, 2024-05-15:
I can definetely say there is something going on with the sequences, even on a clean configuration.

I've defined sequence 1 like this (I also have an action on the button that calls it): 
[image]

In the console I see this:
```

[171842] Pedal/Button: 6/1     EventType: Release           ButtonState: 1
[172242] Pedal/Button: 6/1     EventType: Click             ButtonState: 1
[173268] Pedal/Button: 6/1     EventType: Press             ButtonState: 0
Action ....
=======================================================
SEQUENCE.....Number  1
-------------------------------------------------------
=======================================================
Action ....BANK+.....7
[173427] Pedal/Button: 6/1     EventType: Release           ButtonState: 1
[173827] Pedal/Button: 6/1     EventType: Click             ButtonState: 1
```

Then I set sequence 2 like sequence 1 and suddently in the console I got this:

```
[334659] Pedal/Button: 6/1     EventType: Press             ButtonState: 0
Action ....
=======================================================
SEQUENCE.....Number  1
-------------------------------------------------------
LED COLOR.....Led  5......RGB Color #0a0000
=======================================================
Action ....BANK+.....8
[334812] Pedal/Button: 6/1     EventType: Release           ButtonState: 1
[335212] Pedal/Button: 6/1     EventType: Click             ButtonState: 1
```

So it seems it calls sequence 2, not 1.

raidolo, 2024-05-16:
> Try version 3.3.4

Fixed! 

Even though now in the console it prints the sequence numbers starting by 0

```
[84602] Pedal/Button: 1/1     EventType: LongPress         ButtonState: 0
Action ....BANK-.....20
Action ....
=======================================================
SEQUENCE.....Number  0
-------------------------------------------------------
LED COLOR.....Led  6......RGB Color #0a0009
LED COLOR.....Led  5......RGB Color #000000
LED COLOR.....Led  1......RGB Color #000000
LED COLOR.....Led  2......RGB Color #000000
LED COLOR.....Led  3......RGB Color #000000
LED COLOR.....Led  4......RGB Color #000000
LED COLOR.....Led  7......RGB Color #000000
LED COLOR.....Led  8......RGB Color #000000
LED COLOR.....Led  9......RGB Color #000000
LED COLOR.....Led 10......RGB Color #000000
=======================================================
[85102] Pedal/Button: 1/1     EventType: RepeatPressed     ButtonState: 0
```

alf45tar, 2024-05-16:
Fixed the log too.

closed 2024-05-16 by alf45tar