alf45tar/PedalinoMini #512: Navigate banks + program change

aFunkyBass, 2024-05-09:
Hello everyone, I've been struggling with this for a while, maybe I'm missing something...

What's the best configuration to navigate setups? I have a few bedalboards configured on my MODEP system and I'd like to navigate them along with the associated bank on Pedalino. Every bank on Pedalino is named according to the pedalboard name.

For now I have configured two repeated press switches on the global bank to browse the setups like so:

[image]

Either the program change msg and the bank change are issued correctly, but the display show the generic program change screen instead of the bank name. Obviously it would be more convenient to scroll the banks using their names not their number. This is what I see when browsing

[image]

Is there any better option to scroll between banks and program changes?

Thank you.

raidolo, 2024-05-11:
@aFunkyBass  It depends of the action order which tag is displayed.
If you put the bank increments before the program change and you use a progam change name with the colon at the beginning (meaning "do not switch display), you will have just the next bank name when you switch, also the bank name should start with a colon as well:

[image]

[image]

raidolo, 2024-05-11:
@aFunkyBass I'm using MOD software as well on a mini pc, not modep but their original version running in linux x86. You are not using snapshots? Do you prefer to have midi commands to activate the single pedals when you need them? I'm just trying to figure out which is the best way to configure the Pedalino with MOD software... thanks in advance.

aFunkyBass, 2024-05-12:
> @aFunkyBass It depends of the action order which tag is displayed. If you put the bank increments before the program change and you use a progam change name with the colon at the beginning (meaning [...]

Ciao Alessio,

I think I tried this options during my test, before posting my issue but i didn't work. I probably went wrong at some point, but I'll give it another try following you screenshot example. Thank you.

aFunkyBass, 2024-05-12:
> @aFunkyBass I'm using MOD software as well on a mini pc, not modep but their original version running in linux x86. You are not using snapshots? Do you prefer to have midi commands to activate the [...]

That's cool. I haven't even thought about using a mini PC with the MOD software. Ho did you install it, is there a binary build to make it run on x86 or you compiled it yourself?

As for the snapshots they are a quite handy solution, but as a bassist I guess it's a little bit of an overkill. I don't need dramatically change of tone during the performance as guitarists do. I mostly use a clean sound adding some effects from time to time, so I guess ultimately the "stompbox mode" is the more appropriate. Plus, I happen to play with a lot of different bands and artists so I seldom have a predefined setlist and I cannot pre-organize sounds on a per-song basis.

I sometimes happen to play in situations in which I have to create purely sounds effects emulating keyobards, creating pads and so on... Those kind of effects chain cannot be included in my standard live gig set, that's why I was looking for a solution to switch pedalboards instead.

Thank you for your advices.
Guido.

P.S.: sei italiano? :)

raidolo, 2024-05-12:
> That's cool. I haven't even thought about using a mini PC with the MOD software. Ho did you install it, is there a binary build to make it run on x86 or you compiled it yourself?
> 
> 
> Thank you for your advices.
> 
> Guido.
> 
> 
> 
> P.S.: sei italiano? :)

I should have known you are a bassist looking at your nickname 😂

Regarding MOD software, I'm using a docker image compiled and fixed by me running in Linux Debian, you can have a look at my repository here on github, it's called "mod-docker", but recently they released MOD-Desktop which runs in windows, Mac and Linux. So you could try it  wherever you want. The pedalboards midi switching does not work yet (that's what I've fixed in mine, not like they did on MODEP, but pretty close).

Yes I'm Italian 👍