alf45tar/PedalinoMini #514: Analog input (expression pedal) is jumpy, need help and possible enhancement

raidolo, 2024-05-16:
Hi @alf45tar 

I've rebuilt my Pedalino-8 with the Lilygo T-Display S3, this time I've bought a M-Audio expression pedal and I've added a TRS jack to connect it to the esp board.
Unfortunately and with my big surprise, even if I've added pull-up resistors, a capacitor of 100uF on both input and power, the output is very unreliable, no matter if I use the USB power or the battery. The firmware outputs the correct value but then each second it outputs the max value of 127, when the pedal is full on tip, when the pedal is full toe it outputs half the max value. I saw this in the console and using a midi monitor program. In the console i see a weird "input", with crazy high value in between the right readings from the pedal.
I've tried also to use different gpios without any luck. Moreover, because I'm using 8 foot switches as momentary 2, plus a Midi Out and a TRS jack, I don't have enough gpios to try other solutions. I've also played with the Ease settings and the threshold without any luck.

1) is there any solution to this behavior? How do you connect the expression pedal if you use any?

2) would a possible enhancement in the firmware the ability to use an i2c ADS converter like the ADS1115 and use the i2c bus to do all the analog reads? It supports 4 analog inputs. As far as I know it should be way less prone to bad readings coming from the crappy ADC converters in the esp32 boards...

Thanks in advance.

alf45tar, 2024-05-17:
Can you post some shots, the console log, the pedal configurations?

alf45tar, 2024-05-17:
I ordered an ADS1115 for testing.

raidolo, 2024-05-18:
> Can you post some shots, the console log, the pedal configurations?

this is what I got in the console: 

```
CONTROL CHANGE.....Code   0......Value  89.....Channel  1
Pedal  5   input 779 output 733 velocity 1.37
Action:
CONTROL CHANGE.....Code   0......Value  91.....Channel  1
Pedal  5   input 771 output 723 velocity -10.62
Action: 
CONTROL CHANGE.....Code   0......Value  90.....Channel  1
Pedal  5   input 3091 output 1023 velocity 0.78
Action:
CONTROL CHANGE.....Code   0......Value 127.....Channel  1
Pedal  5   input 770 output 722 velocity -650.11
Action: 
CONTROL CHANGE.....Code   0......Value  90.....Channel  1
Pedal  5   input 3085 output 1023 velocity 0.30                       <<<<<<<<<< THIS INPUT
Action:
CONTROL CHANGE.....Code   0......Value 127.....Channel  1       <<<<<<<<<< THIS OUTPUT
Pedal  5   input 771 output 723 velocity -634.25
Action:
CONTROL CHANGE.....Code   0......Value  90.....Channel  1
Pedal  5   input 770 output 722 velocity -2.61
Action: 
CONTROL CHANGE.....Code   0......Value  90.....Channel  1
Pedal  5   input 3085 output 1023 velocity 0.30
```

I'm using D6 and A6 for the analog pedal (GPIO1 and 18), D5 is the MIDI_OUT on GPIO21. This is the PEDALS config:

```
#elif defined ARDUINO_LILYGO_T_DISPLAY_S3   // https://github.com/Xinyuan-LilyGO/T-Display-S3
#undef  PEDALS
#define PEDALS                7
// https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/gpio.html
// Strapping pins: 1, 3, 45, 46
// ADC1:    1-10
// ADC2:   11-20 (19=D- 20=D+) (ADC2 module is also used by the Wi-Fi)
// SPI0/1: 26-32 (usually used for SPI flash and PSRAM)
// SPI0/1: 33-37 (board with ESP32-S3R8 chip not recommended for other uses)
const byte pinD[] = {GPIO_NUM_10, GPIO_NUM_2,  GPIO_NUM_43, GPIO_NUM_44, GPIO_NUM_1, GPIO_NUM_14, GPIO_NUM_0};
const byte pinA[] = {GPIO_NUM_11, GPIO_NUM_12, GPIO_NUM_13, GPIO_NUM_16, GPIO_NUM_18, GPIO_NUM_14, GPIO_NUM_0};
#define FACTORY_DEFAULT_PIN   GPIO_NUM_0    // Button BOOT
#define DIN_MIDI_IN_PIN       GPIO_NUM_1
#define DIN_MIDI_OUT_PIN      GPIO_NUM_21
#define BATTERY_PIN           GPIO_NUM_4    // Pin connected to BAT (BAT is not VBAT)
#define FASTLEDS_DATA_PIN     GPIO_NUM_3
#else
```

The controls are like "Control 5" --> Pedal 5 - Button empty
The Pedal is set to Analog, and I've tried different settings, no matter what, every second I got that weird input...

alf45tar, 2024-06-03:
ADS1115 support is coming

alf45tar, 2024-06-03:
Version 3.4.0 is out.

ADS1115 board connections:

- VDD     -> 3.3V
- GND    -> GND
- SCL     -> PIN_Ax
- SDA    -> PIN_Dx
- ADDR -> GND

raidolo, 2024-06-03:
Hi @alf45tar  Very Cool!! I'm gonna buy one to test it. I have the T-Display S3. Where do you define the Pins for SCL and SDA in the T-Display S3 part in Pedalino.h?

alf45tar, 2024-06-03:
> Hi @alf45tar Very Cool!! I'm gonna buy one to test it. I have the T-Display S3. Where do you define the Pins for SCL and SDA in the T-Display S3 part in Pedalino.h?

It is the digital and the analog pin of the selected pedal. For example if you are selecting the Mode "Analog 4" for Pedal 1, SDA=10 SCL=11

raidolo, 2024-06-04:
> > Hi @alf45tar Very Cool!! I'm gonna buy one to test it. I have the T-Display S3. Where do you define the Pins for SCL and SDA in the T-Display S3 part in Pedalino.h?
> 
> It is the digital and the analog pin of the selected pedal. For example if you are selecting the Mode "Analog 4" for Pedal 1, SDA=10 SCL=11. At the moment you can connect only one ADS1115 but it can be expanded up to 4 ADS1115 for a total of 16 analog inputs.

Ok, I'm soldering it, How have you connected the analog source? Considering that now we have just 1 analog input per analog source? I guess it's GND + TIP of the TRS jack connected on the A0 pin of the ADS1115, right?

alf45tar, 2024-06-04:
Good point. Analog pin is usually connected to RING. We do not have a digital ping for TIP. The port is not like the other.
I think I have to redesign the connections using separate pin for SDA and SCL.

raidolo, 2024-06-04:
> Good point. Analog pin is usually connected to RING. We do not have a digital ping for TIP. The port is not like the others. I think I have to redesign the connections using separate pins for SDA and SCL.

Hi @alf45tar,

I'm having a bit of trouble getting the ADS1115 recognized.

I got this in the console as soon I set the "Analog 4", also no buttons seems to work properly when the i2c bus acts like this:

```
[ 51676][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 53678][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 55680][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 57682][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 59684][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 61686][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
```
I have the SCL on GPIO_18 (A5) and SDA on GPIO_1 (D5), ADDR to GND.

Do you have pull up resistors on SCL and SDA? I'm using the T-Display S3

Regarding the TIP/RING question, yeah I found out... It's the RING on A0. Thanks.

I'm looking for an i2c scanner, to see if I can see anything.

alf45tar, 2024-06-04:
Pull up resistor are in the board.
Did you connected VDD and GND too?

raidolo, 2024-06-04:
> Pull up resistor are in the board. Did you connected VDD and GND too?

Ok so.... 

I've tested a simple i2c scan sketch using SCL and SDA on GPIO_18 and GPIO_1, I did't touch the wiring, and it seems working:

```
23:11:51.246 -> Scanning...
23:11:51.292 -> I2C device found at address 0x48
23:11:51.292 -> done
```
Now I'm going to test the ADS1115 library.

raidolo, 2024-06-04:
The ADS1115 works properly with a simple sketch measuring the voltage:

The Sketch: 

```
#include <Wire.h>
#include <Adafruit_ADS1X15.h>

Adafruit_ADS1115 ads;

float Voltage = 0.0;

#define SDA_PIN 1
#define SCL_PIN 18

void setup(void) 
{
Serial.begin(115200); 
Wire.begin(SDA_PIN, SCL_PIN);
ads.begin(0x48);
}

void loop(void) 
{
int16_t adc0;

adc0 = ads.readADC_SingleEnded(0);
Voltage = (adc0 * 0.1875)/1000;

Serial.print("AIN0: "); 
Serial.print(adc0);
Serial.print("\tVoltage: ");
Serial.println(Voltage, 7); 
Serial.println();

delay(1000);
}
```

This is the Serial output with me slowing moving the pedal connected through the TRS jack:

```
23:33:37.099 -> AIN0: 17439	Voltage: 3.2698126
23:33:37.099 -> 
23:33:38.114 -> AIN0: 17439	Voltage: 3.2698126
23:33:38.114 -> 
23:33:39.109 -> AIN0: 14408	Voltage: 2.7014999
23:33:39.109 -> 
23:33:40.143 -> AIN0: 3347	Voltage: 0.6275625
23:33:40.143 -> 
23:33:41.147 -> AIN0: 3346	Voltage: 0.6273750
23:33:41.147 -> 
23:33:42.140 -> AIN0: 7129	Voltage: 1.3366874
23:33:42.140 -> 
23:33:43.164 -> AIN0: 9351	Voltage: 1.7533125
23:33:43.164 -> 
23:33:44.187 -> AIN0: 12005	Voltage: 2.2509375
23:33:44.187 -> 
23:33:45.181 -> AIN0: 13604	Voltage: 2.5507500
23:33:45.181 -> 
23:33:46.199 -> AIN0: 16559	Voltage: 3.1048124
23:33:46.199 -> 
23:33:47.192 -> AIN0: 17439	Voltage: 3.2698126
23:33:47.192 -> 
23:33:48.184 -> AIN0: 17439	Voltage: 3.2698126
23:33:48.184 -> 
23:33:49.238 -> AIN0: 17438	Voltage: 3.2696249
23:33:49.238 -> 
23:33:50.251 -> AIN0: 17439	Voltage: 3.2698126
```

This is the Pedalino.h I have:

```
#elif defined ARDUINO_LILYGO_T_DISPLAY_S3   // https://github.com/Xinyuan-LilyGO/T-Display-S3
#undef  PEDALS
#define PEDALS                7
// https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/gpio.html
// Strapping pins: 1, 3, 45, 46
// ADC1:    1-10
// ADC2:   11-20 (19=D- 20=D+) (ADC2 module is also used by the Wi-Fi)
// SPI0/1: 26-32 (usually used for SPI flash and PSRAM)
// SPI0/1: 33-37 (board with ESP32-S3R8 chip not recommended for other uses)
const byte pinD[] = {GPIO_NUM_10, GPIO_NUM_2,  GPIO_NUM_43, GPIO_NUM_44, GPIO_NUM_1, GPIO_NUM_14, GPIO_NUM_0};
const byte pinA[] = {GPIO_NUM_11, GPIO_NUM_12, GPIO_NUM_13, GPIO_NUM_16, GPIO_NUM_18, GPIO_NUM_14, GPIO_NUM_0};
#define FACTORY_DEFAULT_PIN   GPIO_NUM_0    // Button BOOT
#define DIN_MIDI_IN_PIN       GPIO_NUM_1
#define DIN_MIDI_OUT_PIN      GPIO_NUM_21
#define BATTERY_PIN           GPIO_NUM_4    // Pin connected to BAT (BAT is not VBAT)
#define FASTLEDS_DATA_PIN     GPIO_NUM_3

```

What could I debug in Pedalino to see what's going on and why I got the i2c error??

Thanks in advance.

alf45tar, 2024-06-05:
Not yet tested on T-Display-S3.
Try to change pin combination because GPIO_NUM_1 is used also for DIN_MIDI_IN.

raidolo, 2024-06-05:
> Not yet tested on T-Display-S3. Try to change pin combination because GPIO_NUM_1 is used also for DIN_MIDI_IN.

I've changed the DIN_MIDI_IN to GPIO_17 which is empty at the moment....

I've tracked another couple of logs in the console now.
As soon I change the PEDAL 5 to ANALOG  4, I have these errors:

```
Pedal  5        1   ANALOG4            PRESS_1       POLARITY+      Channel  1[130738][E][Wire.cpp:137] setPins(): bus already initialized. change pins only when not.          <<<<<<<<<<<<<<<<<< THIS
[130740][W][Wire.cpp:301] begin(): Bus already started in Master Mode.    <<<<<<<<<<<<<<<THIS
[134196][W][AsyncTCP.cpp:1010] _poll(): rx timeout 4
   I2C 0x48
Pedal  6        1   MOMENTARY1         PRESS_1       POLARITY+      Channel  1   Pin D14
Pedal  7      12L   MOMENTARY1         PRESS_1_2_L   POLARITY+      Channel  1   Pin D0
MIDI/MTC Clock Off
... configuration loaded.
[136752][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263   <<<<<<<<<<<<< THEN THIS
[138754][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[140756][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[142758][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[144760][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
```

alf45tar, 2024-06-05:
How many pedals are configured as Analog 4?

raidolo, 2024-06-05:
> How many pedals are configured as Analog 4? Once you have only 1 pedal as Analog 4 restart the board.

Only Pedal 5.... 

I've got something but the I got again the error 263. 

```
Pedal  5        1   ANALOG4            PRESS_1       POLARITY+      Channel  1[ 27779][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
   I2C 0x48
Pedal  6        1   MOMENTARY1         PRESS_1       POLARITY+      Channel  1   Pin D14
Pedal  7      12L   MOMENTARY1         PRESS_1_2_L   POLARITY+      Channel  1   Pin D0
MIDI/MTC Clock Off
... configuration loaded.
Pedal  5   Button  1   input 75 output 0 velocity -0.00
Pedal  5   Button  2   input 77 output 0 velocity -0.00
Pedal  5   Button  3   input 72 output 0 velocity -999.61
Pedal  5   Button  4   input 75 output 0 velocity -1.00
[ 31230][W][AsyncTCP.cpp:1010] _poll(): rx timeout 4
Pedal  5   Button  1   input 315 output 266 velocity 0.01
Pedal  5   Button  1   input 117 output 19 velocity -5.68
Pedal  5   Button  1   input 370 output 335 velocity 7.28
Pedal  5   Button  1   input 407 output 363 velocity 0.64
Pedal  5   Button  1   input 407 output 368 velocity 0.12
Pedal  5   Button  1   input 407 output 371 velocity 0.07
Pedal  5   Button  1   input 255 output 191 velocity -1.40
Pedal  5   Button  1   input 373 output 339 velocity 1.72
Pedal  5   Button  1   input 407 output 363 velocity 0.55
Pedal  5   Button  1   input 407 output 368 velocity 0.11
Pedal  5   Button  1   input 407 output 371 velocity 0.07
Pedal  5   Button  1   input 407 output 373 velocity 0.05
Pedal  5   Button  1   input 407 output 374 velocity 0.02
Pedal  5   Button  1   input 407 output 375 velocity 0.01
Pedal  5   Button  1   input 407 output 376 velocity 0.02
Pedal  5   Button  1   input 357 output 334 velocity -0.01
Pedal  5   Button  1   input 317 output 282 velocity -1.20
Pedal  5   Button  1   input 263 output 209 velocity -1.69
Pedal  5   Button  1   input 239 output 188 velocity -0.49
Pedal  5   Button  1   input 231 output 176 velocity -0.28
Pedal  5   Button  1   input 222 output 165 velocity -0.25
Pedal  5   Button  1   input 207 output 148 velocity -0.39
Pedal  5   Button  1   input 189 output 126 velocity -0.50
Pedal  5   Button  1   input 171 output 103 velocity -0.54
Pedal  5   Button  1   input 145 output 70 velocity -0.76
Pedal  5   Button  1   input 119 output 38 velocity -0.74
Pedal  5   Button  1   input 101 output 9 velocity -0.67
Pedal  5   Button  1   input 83 output 4 velocity -0.12
Pedal  5   Button  1   input 78 output 1 velocity -0.07
Pedal  5   Button  1   input 78 output 0 velocity -0.02
Pedal  5   Button  1   input 128 output 15 velocity 0.01
Pedal  5   Button  1   input 173 output 77 velocity 1.44
Pedal  5   Button  1   input 227 output 146 velocity 1.59
Pedal  5   Button  1   input 283 output 216 velocity 1.60
Pedal  5   Button  1   input 340 output 288 velocity 1.67
Pedal  5   Button  1   input 385 output 338 velocity 1.15
Pedal  5   Button  1   input 407 output 363 velocity 0.58
Pedal  5   Button  1   input 407 output 368 velocity 0.11
Pedal  5   Button  1   input 407 output 371 velocity 0.07
Pedal  5   Button  1   input 407 output 372 velocity 0.02
Pedal  5   Button  1   input 407 output 374 velocity 0.05
Pedal  5   Button  1   input 407 output 375 velocity 0.01
Pedal  5   Button  1   input 407 output 376 velocity 0.02
Pedal  5   Button  1   input 322 output 275 velocity -0.12
Pedal  5   Button  1   input 235 output 166 velocity -2.54
Pedal  5   Button  1   input 166 output 86 velocity -1.85
Pedal  5   Button  1   input 113 output 25 velocity -1.40
Pedal  5   Button  1   input 78 output 9 velocity -0.37
Pedal  5   Button  1   input 78 output 4 velocity -0.12
Pedal  5   Button  1   input 78 output 1 velocity -0.07
Pedal  5   Button  1   input 78 output 0 velocity -0.02
Pedal  5   Button  1   input 138 output 27 velocity 0.03
[... 33 lines cut ...]
Pedal  5   Button  1   input 314 output 265 velocity 2.54
Pedal  5   Button  1   input 386 output 350 velocity 1.98
Pedal  5   Button  1   input 407 output 364 velocity 0.32
Pedal  5   Button  1   input 407 output 369 velocity 0.12
Pedal  5   Button  1   input 407 output 371 velocity 0.05
Pedal  5   Button  1   input 407 output 372 velocity 0.02
Pedal  5   Button  1   input 203 output 126 velocity -2.86
Pedal  5   Button  1   input 97 output 0 velocity -2.95
Pedal  5   Button  1   input 205 output 129 velocity 0.43
Pedal  5   Button  1   input 270 output 201 velocity 1.68
Pedal  5   Button  1   input 196 output 124 velocity -0.89
Pedal  5   Button  1   input 112 output 12 velocity -2.57
Pedal  5   Button  1   input 78 output 5 velocity -0.16
Pedal  5   Button  1   input 78 output 2 velocity -0.07
Pedal  5   Button  1   input 78 output 0 velocity -0.05
Pedal  5   Button  1   input 159 output 58 velocity 0.67
Pedal  5   Button  1   input 242 output 175 velocity 2.72
Pedal  5   Button  1   input 292 output 222 velocity 1.09
Pedal  5   Button  1   input 297 output 229 velocity 0.16
Pedal  5   Button  1   input 249 output 201 velocity -0.66
Pedal  5   Button  1   input 153 output 64 velocity -3.14
Pedal  5   Button  1   input 91 output 3 velocity -1.42
Pedal  5   Button  1   input 78 output 1 velocity -0.05
Pedal  5   Button  1   input 78 output 0 velocity -0.02
```

alf45tar, 2024-06-05:
Can you give a try to latest version 3.4.1?

raidolo, 2024-06-05:
> Can you give a try to latest version 3.4.1?

Yeah sure... meanwhile I was trying to debug the issue, I believe there is something in the refresh_analog4() that's not working....

raidolo, 2024-06-05:
> Can you give a try to latest version 3.4.1?

Ok, now it's better!
I've noticed that because we would have 4 inputs, I should set the Control to Pedal 5 - Button 1, I've seen in the logs that 4 buttons are logged.
Now... It's better, I see the Control Change and the bar in the display, but at some point while I do up and down with the physical pedal, the board hangs and reboots. And also while it's in that state, it boot loops few times.

There's nothing useful in console.

We are getting there but there should be something else that is acting weird.

raidolo, 2024-06-05:
When in hang, meaning "if I move the pedal nothing happens in the display or console", If I don't touch anything and the board does not reboot by itself, it happen one time that it recovered and the I was able to use the pedal again. 
It seems like i2c bus overload or something like that, or an electrical problem. 
Should I try to pull-up the two GPIOs?

raidolo, 2024-06-05:
Ok I had the Activity Threshold and the Ease at the lowest level to test the inputs, it's definetly too much. Now I have Ease 2 Threshold 16, and it seems more stable.

raidolo, 2024-06-05:
I'm having some weird behaviors when I enable ANALOG 4 in the web ui... I got a board reset but I can't trace anything in the console unfortunately :( 
Also, when enabled, the WiFi started to acted strange and sometimes it doesn't connect properly 
I got this in the console at some point:

```
[ 36737][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 8 - ASSOC_LEAVE
SYSTEM_EVENT_STA_DISCONNECTED
```

alf45tar, 2024-08-13:
The upcoming release 3.5.0 will solve the analog input for T-Display S3 board. It solves pins assignment and a software bug.

closed 2026-11-02 by alf45tar