alf45tar/PedalinoMini #518: Can't build tag 3.3.4 lilygo-t-display-s3

zavorra, 2024-06-01:
Hi, 

build of tag 3.3.4 for lilygo-t-display-s3 fails, essentially because of 

`.pio/libdeps/lilygo-t-display-s3/OSC/OSCData.h:65:16: error: 'struct midi' redeclared as different kind of symbol`

full log below

```

 *  Executing task: platformio run 

Processing lilygo-t-display-s3 (platform: espressif32@^6.0.0; framework: arduino; board: lilygo-t-display-s3)
--------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lilygo-t-display-s3.html
PLATFORM: Espressif 32 (6.6.0+sha.2fd630e) > LilyGo T-Display-S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20015.0 (2.0.15) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 55 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AceButton @ 1.10.1
|-- ArduinoJson @ 7.0.4
|-- AsyncTCP-esphome @ 2.1.3
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- FastLED @ 3.7.0
|-- HC-SR04 @ 1.1.2
|-- Improv @ 1.2.3
|-- MD_REncoder @ 1.0.1
|-- NimBLE-Arduino @ 1.4.1
|-- OSC @ 3.5.8+sha.debf01b
|-- ResponsiveAnalogRead @ 1.2.1
|-- TFT_eSPI @ 2.5.33
|-- MIDI Library @ 5.0.2
|-- AppleMIDI @ 3.3.0
|-- BLE-MIDI @ 2.2.0
|-- ipMIDI @ 1.0.3
|-- Hello Drum @ 0.7.7
|-- Adafruit TinyUSB Library @ 2.4.1
|-- WiFi @ 2.0.0
|-- Ticker @ 2.0.0
|-- Preferences @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- SPI @ 2.0.0
|-- Wire @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- Update @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ESP32 Async UDP @ 2.0.0
|-- FS @ 2.0.0
Building in release mode
Compiling .pio/build/lilygo-t-display-s3/src/PedalinoMini.cpp.o
<command-line>: warning: "ARDUINO_USB_MODE" redefined
<command-line>: note: this is the location of the previous definition
In file included from .pio/libdeps/lilygo-t-display-s3/TFT_eSPI/TFT_eSPI.h:97,
                 from src/DisplayTFT.h:13,
                 from src/PedalinoMini.cpp:58:
.pio/libdeps/lilygo-t-display-s3/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:110:3: warning: #warning >>>>------>> DMA is not supported in parallel mode [-Wcpp]
  #warning >>>>------>> DMA is not supported in parallel mode
   ^~~~~~~
[... 191 lines cut ...]
<command-line>: note: this is the location of the previous definition
Compiling .pio/build/lilygo-t-display-s3/lib2ec/NimBLE-Arduino/nimble/nimble/host/mesh/src/friend.c.o
Compiling .pio/build/lilygo-t-display-s3/lib2ec/NimBLE-Arduino/nimble/nimble/host/mesh/src/glue.c.o
<command-line>: warning: "ARDUINO_USB_MODE" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "ARDUINO_USB_MODE" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pio/build/lilygo-t-display-s3/lib2ec/NimBLE-Arduino/nimble/nimble/host/mesh/src/health_cli.c.o
<command-line>: warning: "ARDUINO_USB_MODE" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pio/build/lilygo-t-display-s3/lib2ec/NimBLE-Arduino/nimble/nimble/host/mesh/src/health_srv.c.o
*** [.pio/build/lilygo-t-display-s3/src/PedalinoMini.cpp.o] Error 1
<command-line>: warning: "ARDUINO_USB_MODE" redefined
<command-line>: note: this is the location of the previous definition
======================================================== [FAILED] Took 8.33 seconds ========================================================

Environment          Status    Duration
-------------------  --------  ------------
lilygo-t-display-s3  FAILED    00:00:08.330
================================================== 1 failed, 0 succeeded in 00:00:08.330 ==================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```

zavorra, 2024-06-01:
Also, could you consider adding to the README how to flash the official .bin file locally instead of relying on an online tool?

alf45tar, 2024-06-01:
It is an issue with the latest changes in the linked library.
Replace in `platform.ini`

`https://github.com/CNMAT/OSC@^3.5.8`

with

`https://github.com/CNMAT/OSC.git#3.5.8`

alf45tar, 2024-06-01:
> Also, could you consider adding to the README how to flash the official .bin file locally instead of relying on an online tool?

Use `Verbose Upload` under `Advanced` in Platformio to see the command used.

It should similar to this

`"/Users/alf45tar/.platformio/penv/bin/python" "/Users/alf45tar/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32s3 --port "/dev/cu.usbmodem14201" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0000 /var/folders/3_/qh1y3s4x79g4rh4_gk92stqm0000gn/T/PedalinoMini/build/lilygo-t-display-s3/bootloader.bin 0x8000 /var/folders/3_/qh1y3s4x79g4rh4_gk92stqm0000gn/T/PedalinoMini/build/lilygo-t-display-s3/partitions.bin 0xe000 /Users/alf45tar/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin 0x10000 /var/folders/3_/qh1y3s4x79g4rh4_gk92stqm0000gn/T/PedalinoMini/build/lilygo-t-display-s3/firmware.bin`

zavorra, 2024-06-01:
> Use `Verbose Upload` under `Advanced` in Platformio to see the command used.
>
> It should similar to this
>
> `"/Users/alf45tar/.platformio/penv/bin/python" [...]

Thanks for both. I think this  can be closed 

It seems that, since the "official" .bin files (those in the firmware subdir) are already composed of bootloadr+partitions+boot_app+firmware all I nedded was something similar to 

`/home/giorgio/.platformio/penv/bin/python "/home/giorgio/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32s3 --port /dev/ttyACM* --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0000 firmware/lilygo-t-display-s3/lilygo-t-display-s3-3.2.4-firmware.bin `

Writing here for the records..

closed 2024-06-01 by alf45tar

italocjs, 2024-07-19:
> https://github.com/CNMAT/OSC.git#3.5.8

Doing this allowed me to compile too, i didnt see any PR related, shouldn't there be one so it can be updated on main branch?

alf45tar, 2024-07-19:
The main branch will be updated as per rules here

https://github.com/alf45tar/PedalinoMini?tab=readme-ov-file#sponsors