alf45tar/PedalinoMini #522: LilyGo T-Display S-3 3.2.2 Web Login Issue

NewmarchP, 2024-07-24:
Hi,

I'm really looking forward to working with this project. I have previously built my own, much simpler, pedalboard solution for Arduino and I'm really looking forward to working on this one.

I have installed the firmware and configured WiFi. When I attempt to access the Web User Interface I am asked for Username & Password. I saw in another thread this was resolved with some admin credentials and an acknowledgement to remove the credential requirement for first login. I tried these, no luck. I've tried everything else I can think of, including re-installing 3.2.1, however everytime I try to re-install it times out.

Any advice for how to proceed?

Thanks again, I love the look of this project and will be sharing it with other musos once I have it up and running.
Pete

regisampa, 2024-07-24:
Try to locate your device via bluetooth, the password will be this number that you located via bluetooth, example "C88854512" and the user is "admin".

NewmarchP, 2024-07-24:
Well, that worked. I can't believe I didn't try that. Thanks for your help!

alf45tar, 2024-07-25:
Device name, which is the default password for old firmwares, is displayed pressing the BOOT button too.

closed 2024-07-25 by alf45tar