alf45tar/PedalinoMini #523: MIDI DIN for LilyGo T-Display S3

NewmarchP, 2024-07-27:
Hi,

Love this project, I've got 3.4.2 running well and all my switches are working!

The last hurdle I can see is MIDI OUT via DIN connection. I tested it with the schematic in the README, but no luck.
I found another Issue log which detailed switching resistor values for 3.3V rather than 5V. Did that, still no luck.

I looked further into the Pedalino.h document to recheck the Pin values. Turns out I had MIDI OUT off Pin 3 as per 3.2.2. In 3.4.2 it looks like it should be on Pin 1. But there is also a switch on Pin 1, which I have attached.

Do I replace the switch for the MIDI OUT tx in Pin 1 and then relocate my switch to 14?

Thanks for the help so far. I've built lots of analog guitar pedals, but I've only done a handful of Arduino projects, so feeling a little out of my comfort zone.

Pete

alf45tar, 2024-07-27:
There are not enough available pins on LilyGo T-Display S3 board to implement all.
I decided to not support it on LilyGO T-Display S3 board because the board is USB MIDI capable.
If you want to implement it you have to find an available pin for example reducing the PEDALS to 7.

NewmarchP, 2024-07-27:
That makes sense, sorry for the beginner questions and thanks for your help. 

Just to confirm, in VSCode I need to edit pedalino.h to remove the 8th pin and add the DIN MIDI OUT pin as the removed pin. Sound about right? Sorry, working on my phone at a gig, I was SO CLOSE to having the switcher ready. 

Thanks again, I think I'm going to love this thing!
Pete

closed 2024-07-29 by NewmarchP