alf45tar/PedalinoMini #524: Not able to Build

Wapata, 2024-08-02:
Hi !
I'm confused because it's been a while and i'm not able to build anymore the last firmware. Plus i don't have a clue of how tu push the last Sponsor one.

But... I think that this time the fault is not completely on my side :
```
Processing ttgo-t-display (platform: espressif32@^3.5.0; framework: arduino; board: esp32dev)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------Library Manager: Installing git+https://github.com/CNMAT/OSC @ ^1.3.7
git version 2.41.0.windows.1
Cloning into 'C:\Users\Pierre\.platformio\.cache\tmp\pkg-installing-rm58thby'...
remote: Enumerating objects: 122, done.
remote: Counting objects: 100% (122/122), done.
remote: Compressing objects: 100% (97/97), done.
Receiving objects: 100% (122/122), 103.61 KiB | 1.82 MiB/s, done.
Resolving deltas: 100% (28/28)
Resolving deltas: 100% (28/28), done.
PackageException: Package version 3.5.8+sha.778dd98 doesn't satisfy requirements ^1.3.7 based on PackageMetadata <type=library name=OSC version=3.5.8+sha.778dd98 spec={'owner': None, 'id': None, 'name': 'OSC', 'requirements': '^1.3.7', 'uri': 'git+https://github.com/CNMAT/OSC'}

 *  The terminal process "C:\Users\Pierre\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'ttgo-t-display'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

Here is the question  i've uploaded the "OSC" libraries, and "3.5.8" is upper than "1.3.7" so... what's wrong here ?
thank you for your help ! See ya !

alf45tar, 2024-08-02:
Replace

`https://github.com/CNMAT/OSC@^1.3.7`

with

`https://github.com/CNMAT/OSC@^3.5.8`

in `platformio.ini`

Wapata, 2024-08-02:
[image]

Success !!!
Thanks a lot ! 

How do I jump to 3.4.2 ?
[image]

I think (memory lack) that uploading via the interface do not work with every boards because of space limits ?

alf45tar, 2024-08-03:
Online update works only if the flash memory is double of the firmware size.

closed 2024-08-13 by alf45tar