alf45tar/PedalinoMini #525: Resetting too much

Wapata, 2024-08-02:
Hi !
My board ttgo-t-display do reset a lot while using the web interface.
It's been a year and some firmware since it do that so... Let's say it's hardware :
What do you think about a little filtering cap somewhere?
Do you have the same behavior at first ?

alf45tar, 2024-08-02:
Use the latest version available for sponsors.

Wapata, 2024-08-02:
I don't know how to install it :-$

alf45tar, 2024-08-02:
Follow the instruction here https://github.com/alf45tar/PedalinoMini?tab=readme-ov-file#how-to-build-and-upload
just replace the repository with https://github.com/pedalino-sponsors/PedalinoMini

Wapata, 2024-08-02:
Thanks.
Mmmm... I may have done something wrong, it display the "P" "Wifi" "BT" logo and reboot infinitly and https://alf45tar.github.io/PedalinoMini/installer do not have the "wifi" menu.

alf45tar, 2024-08-02:
Erase the flash and upload firmware and system image.

Wapata, 2024-08-02:
Nice to hear that, because that what i'm doing right now (but have a computer so old that it take some serious time).

It's been a year since the last time i've done that, i'm happy to have some good habits, and that you're still here to kindly help !

Wapata, 2024-08-02:
Done ! (aoutch, 45 minutes !)
Put it back in my wifi, and... It reset at nearly every clic :-/
[image]

Wapata, 2024-08-03:
> Use the latest version available for sponsors.

Hi !
Bad news, I have the same behaviour with BPI-LEAF-S3. 
It start his own wi-fi, and I can see the pages badly as the last screen shot, and it reseat when i navigate (like... to change the wi-fi, i've just clicked on "option" and the board did reset. I've heard the windows sound from my computer and been disconnected form my phone-web-browser.)

alf45tar, 2024-08-03:
Can you confirm the following step?

- Click “Upload File System Image” under Platform. Do not skip this step otherwise the WebUI will not works properly.

Wapata, 2024-08-03:
...
I did clic on the wrong button... I feel shame now.
But in the end it work ! It seem more stable, the TTGO board did reset when uploading or applying the profiles, but in the end, it's the only time right now (and there is a problem because the actions does not come back). Anyway i've made screenshots of the configuration so... i will restart from the begenning.

Thank you, again !

closed 2024-08-03 by Wapata