alf45tar/PedalinoMini #527: Feature request - Filter active sensing message 0xFE

wcraffonara, 2024-08-08:
Hi,
  Would it be possible to implement a simple filter for MIDI active sensing messages (0xFE)? 

Implementation: if input message from MIDI IN is 0xFE (active sensing) then do not send it to MIDI OUT.

Use case: some MIDI controllers like Boss ES8/5 do send continous active sensing MIDI messages which causes unwanted noise/popping in audio signal when audio devices do not implement a good shielding between audio and MIDI interface. This noise can be eliminated by filtering this continous active sensing messages.

Thank you for your support

Best regards
Walter

wcraffonara, 2024-08-31:
Will you implement it? or can you tell me where in the code I can add it?
Thank you for your support.

alf45tar, 2024-08-31:
Search for On*ActiveSensing functions in all files and comment the code.

wcraffonara, 2024-10-07:
Hi,
  I have now commented out all the On*ActiveSensing functions, but the Active Sensing Command is still forwarded from MIDI IN to MIDI OUT. It also does not help if the MIDI THRU option is switched off. What could be the reason for this?

wcraffonara, 2024-10-11:
Hi,
  sorry, my fault. It's not the Active Sensing message I want to filter but I need to filter the MIDI System Realtime Messages. Can you give me a hint what I should modify in the code or were I can find the place were the Realtime Message are sended?

Thank you very much!

Regards
Walter

wcraffonara, 2024-10-11:
Solved: in MIDI.hpp line 1457 comment out "case SystemReset" in funciton void MidiInterface<Transport, Settings, Platform>::thruFilter(Channel inChannel)

closed 2026-11-02 by alf45tar