alf45tar/PedalinoMini #530: wifi installation after update firmware sponosor

mike76700, 2024-08-16:
after updating with platformio firmware 3.4.2.
I did build, upload and upload filesystem image.
when I go to pedalino installer it doesn't suggest wifi connection, only to install with the old version.
here's a screenshot log
[image]

mike76700, 2024-08-16:
did i update firmware the wrong way?

alf45tar, 2024-08-16:
Which version/board did you installed? Your PedalinoMini is booting in BLE mode.

mike76700, 2024-08-16:
esp32doit-devkit-v1

mike76700, 2024-08-16:
for the moment I'm trying to integrate the pedalino into the esp before doing all my assembly around it.
i've got another problem at the moment: when i plug in the esp, the screen shows me this
it's an esp 32s (wroom-32) equivalent to devkit (just a different pinout than 32doit)
[image]

alf45tar, 2024-08-17:
Yes, it works with SSH1106 too. Check at the top of DisplayOLED.h file.

```
/*
__________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___
\______   \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \
 |     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \
 |    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
 |____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  /
               \/     \/     \/             \/               \/        \/       \__\                 \/  /__/
                                                                                   (c) 2018-2024 alf45star
                                                                       https://github.com/alf45tar/PedalinoMini
 */

#include "Fonts.h"

#if defined(HELTEC_WIFI_KIT_32)
#include <oled/SSD1306Wire.h>
#include <oled/OLEDDisplayUi.h>
#define OLED_I2C_ADDRESS  0x3c
#define OLED_I2C_SDA      SDA_OLED
#define OLED_I2C_SCL      SCL_OLED
SSD1306Wire               display(0x3c, SDA_OLED, SCL_OLED, RST_OLED, GEOMETRY_128_64);
#endif
#if defined(TTGO_T_EIGHT)
#include <SH1106Wire.h>
#include <OLEDDisplayUi.h>
#define OLED_I2C_ADDRESS  0x3c
#define OLED_I2C_SDA      21
#define OLED_I2C_SCL      22
SH1106Wire                display(OLED_I2C_ADDRESS, OLED_I2C_SDA, OLED_I2C_SCL);
#endif
#if defined(SSH1106WIRE)
#include <SH1106Wire.h>
#include <OLEDDisplayUi.h>
#define OLED_I2C_ADDRESS  0x3c
#define OLED_I2C_SDA      SDA
#define OLED_I2C_SCL      SCL
SH1106Wire                display(OLED_I2C_ADDRESS, OLED_I2C_SDA, OLED_I2C_SCL);
#endif
#if defined(SSD1306WIRE)
#include <SSD1306Wire.h>
#include <OLEDDisplayUi.h>
#define OLED_I2C_ADDRESS  0x3c
#define OLED_I2C_SDA      SDA
#define OLED_I2C_SCL      SCL
SSD1306Wire               display(OLED_I2C_ADDRESS, OLED_I2C_SDA, OLED_I2C_SCL);
#endif

#define DISPLAY_WIDTH   display.getWidth()
#define DISPLAY_HEIGHT  display.getHeight()
```

alf45tar, 2024-08-17:
> esp32doit-devkit-v1

Did you try to reset to factory default configuration?

mike76700, 2024-08-17:
I managed to get the pedalino to work, I activated the right type of screen in platformio.ini, which also enabled me to boot in normal mode and configure everything correctly, access to the webui is perfect. thanks again for the help.

Designbase, 2024-08-17:
Dank je voor je bericht!

Ik ben met vakantie tot en met 1 September.

De mail wordt gelezen en mochten er calamiteiten zijn dan zal ik actie ondernemen!

Met vriendelijke Groet,

Ronald Westland
Designbase

closed 2024-08-17 by alf45tar