alf45tar/PedalinoMini #533: button hold logic trigger different cc command for expression pedal

quiknick, 2024-09-18:
Did not already see this existing, but would like logic introduced that while I have a a button held down, there will be logic to trigger  a separate cc command for expression pedal to send 0-127 and then when I let go of button, expression pedal returns to regular cc 0-127 as previously defined.

alf45tar, 2024-09-18:
It works only using hands not using feet. Did you considered the simultaneous press feature?

quiknick, 2024-09-18:
Can you explain this, I am not seeing the Simultaneous press feature.

alf45tar, 2024-09-18:
Go to Controls page and set a Control with 2 buttons. The Action using the Control is triggered on simultaneous press of the 2 buttons.

quiknick, 2024-09-18:
2 buttons is not the solution I am looking for here, but that is still useful for bank up/down scenarios. 

I want to have the same expression pedal use a different cc when a button is held down to adjust the value 0-127.

alf45tar, 2024-09-19:
PedalinoMini have similar feature that can be activated using feet too. Check the Repeat or Repeat Overwrite actions.
If you assign Repeat to your expression pedal it will repeat the last CC.

quiknick, 2024-09-19:
That works for last cc, but would mean on Program changes I have to define the default cc for expression pedal on each program change, then on the button hold down define the cc that the expression will use repeat overwrite. Will try this method. Thanks.

quiknick, 2024-09-19:
Ok this will not work. because I am using sequence now. Last cc runs the entire sequence again for repeat overwrite. Even when I put the cc on another sequence and make the previous sequence use the next sequence with only the cc change I want as default.  Can you improve on this or provide another way to accomplish a default cc value on the pedal, but allow for button changes / holds in my case to define a different cc that I can then adjust via the expression pedal?

mknerr, 2024-10-29:
This is a little late, I know, but what if you had a hardwired switch that would physically change the connection of the CC pedal to a different pin. You could use a momentary switch if you wanted to, but I would rather be able to be on/off and do it with one foot.
The Line 6 HD500 has a switch that you can engage when you "toe down" all the the way with the CC Pedal. I can't say if it's physically switching the control input but I suspect it is. It's definitely a "clicky" switch. You can set either mode to control whatever you want, but typically it's always Volume and then you switch it to control the Wah Wah.