alf45tar/PedalinoMini #534: Button hold logic is triggered after button press

quiknick, 2024-09-18:
Button hold logic seems incorrect as button press is triggered before hold. Currently I have program change on button 1, 2 and 3, and separate logic on button holds, but when I button hold, button press triggers and causes program change, when I wanted the button hold logic to work. This is confirmed on console as well.

```
[27306441] Pedal/Button: 1/1     EventType: Press             ButtonState: 0
Action ....=======================================================
SEQUENCE.....Number  1
-------------------------------------------------------
PROGRAM CHANGE.....Program   1.....Channel  3
LED COLOR.....Led  1......RGB Color #000000
CONTROL CHANGE.....Code 102......Value 127.....Channel  3
LED COLOR.....Led  1......RGB Color #000000
=======================================================
[27306941] Pedal/Button: 1/1     EventType: LongPress         ButtonState: 0
[27307441] Pedal/Button: 1/1     EventType: RepeatPressed     ButtonState: 0
```
I recommend checking out library [RoxMux](https://github.com/neroroxxx/RoxMux) which is a great library for your use case for encoders, buttons, lights and midi and etc. 1 library and modules you can enable as desired. I suggest it because you could trim down on other libraries, additionally the logic in the RoxButton portion of RoxMux has the following logic in this order:

1. Button Held
2. Button Double Press
3. Button Press
4. Button Released

```
#include <RoxMux.h>

RoxButton button;

void setup(){
  // for this example pin 0 on my teensy is connected to a momentary button
  pinMode(0, INPUT_PULLUP);
  // pin 13 is the default led it will light up the led when the button is pressed
  pinMode(13, OUTPUT);
  // start the serial monitor and print a line
  Serial.begin(115200);
  Serial.println("RoxButton Example");
  // little delay before starting
  delay(100);
  // the .begin() method starts the debouncing timer
  button.begin();
  // if you will be using double press then you can set the time interval between
  // pressed in milliseconds, in the line below we set it to 300ms which is the
  // default value.
  // the double press is triggered based on the first release and next press of the button
  // that is if the button is currentley in it's open state, and it changes to
  // it's closed state, as soon as it goes back to the open state a flag is set
  // in which case the program will wait for the next time the button goes into
  // it's closed state, as soon as the button goes into the closed state then
  // another check is performed in which case if that 2nd pressed happened in less
  // than 300 milliseconds then that's a double press.
  button.setDoublePressThreshold(300);
}

void loop(){
  // 3 parameters passed to update method

  // #1 the actual state of the button, in this case we passs the output of digitalRead
  //    RoxButton doesn't read the pin itself so you have to pass the state of
  //    the pin, this is because RoxButton is meant to also work with Multiplexer
  //    outputs, so you have to read your pin or mux pin and then give that
  //    state to RoxButton and the class will do the job for you.
  //    ROXBUTTON WILL DEBOUNCE THE READING, DO NO DEBOUNCE IT YOURSELF.

  // #2 (optional) the debounce time, this value is in milliseconds, default: 50ms
  //    this is how long to wait before reading the button again

  // #3 (optional) the state of the button when it's active,
  //    if your pin has a pullup resisitor then the active state will be LOW,
  //    if it's a pulldown resistor then the active state is HIGH.
  button.update(digitalRead(0), 50, LOW);

  // if you want your button to have dual functionality like one for press and one
  // for hold, you will have to use held() and released() not pressed()
  // otherwise the button will always read the pressed() method.

  // the .held() function can take the time in milliseconds that you want to hold
  // the button for, if no value is passed the default 500ms will be used
  if(button.held()){

    Serial.println("* Button has been held for 500 milliseconds");

  } else if(button.doublePressed()){

    // NOTE: when you first press a release a button both the press and release
    // will be triggered, however if you press the button a second time within
    // the default 300ms time for double press, neither the press nor the release
    // will be triggered, only the double press,
    // if you want to trigger the release of the button after a double press
    // you can pass true to the .doublePressed(true) function
    Serial.println("& Button has been double pressed");

  } else if(button.pressed()){

    Serial.println("+ Button has been pressed");

  } else if(button.released(true)){
    // you can pass a boolean (true or false) to the .released() method
    // true if you want to ignore the "released" reading after "held" was triggered
    Serial.println("- Button has been released");
  }
  // if you only want to know if the button was held then
  digitalWrite(13, button.getCurrentState());
}
```

alf45tar, 2024-09-18:
Press action is not triggered when long press is detected. The log includes every event but actions are not triggered.
Long Press and Double Press must be activated on Pedals page.
Press action is postponed until Long Press Time (500 ms by default) defined in Configuration page. If button is released before Long Press Time the Press action is triggered. If button is keep pressed for at least Long Press Time, only the Long Press action is triggered.

quiknick, 2024-09-18:
I am seeing it as a bug, if it is intended to work as described. I am holding down, but regular press is happening sending command as in my log above while still holding prior to the 500ms or any other value defined on config page.

alf45tar, 2024-09-18:
Did you enabled Long Press in Pedals page?

quiknick, 2024-09-18:
[image]

Yes I have enabled Long Press in Pedals page
[image]

alf45tar, 2024-09-18:
Disable Latch Emulation

quiknick, 2024-09-18:
I need the latch emulation as well for press and release functions. Is there a way to include logic that If I press a button It does just the 1st action then if I press later it does the second action rather than the 1st action.

basically I have 3 buttons which on press run a sequence which are program changes  and cc  127 to activate.
I want ability if I am still on that program if I press the button again it will only deactivate so on release I have cc 0 rather than 127. and If I press again it will do the sequence again.

Without the latching function the release is triggered right after pressing/releasing.

Then I want ability to hold button down and do something else.  and also ability to do double press as well.

quiknick, 2024-09-18:
Even with disable latch emulation,
long press is activated after press action. so press action commands are sent first. If I hold down, no commands should be sent but the hold down commands defined.

alf45tar, 2024-09-18:
Long Press on latch has no sense. You can obtain the result in this way.

1. Disable Latch Emulation
2. On Press execute a sequence step by step
3. The first step of a sequence is a second sequence
4. The second step of the first sequence is CC 0
5. Add PC and CC 127 to the second sequence

alf45tar, 2024-09-18:
> Even with disable latch emulation, long press is activated after press action. so press action commands are sent first. If I hold down, no commands should be sent but the hold down commands defined.

Show me the Configuration page

quiknick, 2024-09-18:
I tried your method, but it just runs both sequences so if cc 0 on 1st sequence second step it runs after 2nd sequence of 1st step each time.
sequence 1
[image]
sequence 4
[image]
```
[3101567] Pedal/Button: 1/1     EventType: Press             ButtonState: 0
Action ....=======================================================
SEQUENCE.....Number  1
-------------------------------------------------------
=======================================================
SEQUENCE.....Number  4
-------------------------------------------------------
CONTROL CHANGE.....Code 127......Value   0.....Channel  3
LED COLOR.....Led  1......RGB Color #000000
PROGRAM CHANGE.....Program   1.....Channel  3
=======================================================
CONTROL CHANGE.....Code 102......Value   0.....Channel  3
LED COLOR.....Led  1......RGB Color #000000
=======================================================
[3101722] Pedal/Button: 1/1     EventType: Release           ButtonState: 1
```

I think you misunderstood I already have a sequence set up. I just want when I press the button the next time to trigger bypass by sending cc 0 and the next time the button is pressed it will trigger sequence 1 again.
[image]

[image]

Options currently all default
[image]

alf45tar, 2024-09-18:
On Press select Send -> Step by Step +

quiknick, 2024-09-19:
Awesome. Sorry for not understanding intent of Step by Step+. I now see it was just briefly mentioned in 3.1.5 release notes "Added "Step by Step+" and "Step by Step-" to run sequence step by step forward and backward". Well that solves that logic, but didn't solve the long press issue I still see as a bug where long press triggers after a button press event sending a command prior to long press.

quiknick, 2024-09-19:
I think removing the latching and fixing the controls/sequences solved this issue. I only see the command happen as expected so I am closing this. Thank you for your help.

quiknick, 2024-09-19:
This can be closed

closed 2024-09-19 by alf45tar