alf45tar/PedalinoMini #536: Compiling Only Necessary Features

prostealth, 2024-10-17:
just upload the latest firmware on esp32 devkit but it keep restarting.
then i upload my old firmware v2.2.10 and it can work properly.
i think esp32 cannot handle all these new features.
how can i remove the unnecessary feature on the code?

alf45tar, 2024-10-17:
did you reset PedalinoMini to factory default?

alf45tar, 2024-10-17:
Is display working? Do you see the WIFI and Bluetooth logo?
If yes keep the boot button pressed to reset to factory default.

prostealth, 2024-10-18:
unfortunately im not using display..
does this restarting problem usually happen?
and resetting to factory will fix that?

alf45tar, 2024-10-18:
It depends which is the issue but in most of the cases it is the solution.
Can you attach here you error log?

prostealth, 2024-10-18:
nothing in the log.. its booting but restarting on wifi/smartconfig process..

alf45tar, 2024-10-18:
if you don't have a display how can you say the reboot is on smartconfig?

prostealth, 2024-10-18:
because its restarting in certain seconds.
usually it will change to AP mode if it doesnt found or connect to a network.
i think i will test it further. but with new board and lcd too.

closed 2025-07-25 by alf45tar