alf45tar/PedalinoMini #538: Tap Tempo LED

Pedalnetics, 2024-11-10:
I saw references to an LED responding at the Tap Tempo BPM in https://github.com/alf45tar/PedalinoMini/issues/104#issuecomment-671795844, however I haven't been able to find any of the mentioned places/ability to set an LED to respond @ the BPM.

Any guidance/redirection for the newer firmware of where that ability is at?

alf45tar, 2024-11-10:
It works only when PedalinoMini is a MIDI Clock Master mode. Define a pedal actions to enter in MIDI Clock Master mode.

Pedalnetics, 2024-11-10:
[image]
I have a single footswitch right now programmed for MIDI Clock Master, and a second footswitch programmed for Tap, and it does load to the MIDI clock screen showing Master, BPM, and time signature, and I'm able to tap in a BPM properly; However no LED's are lighting.

If I change over to Set LED Color command, or even a CC command, and set the LED color that way, the lights do properly turn on even when I leave the LED Number to Default (I have them mapped properly under Controls); and another bank using Control Change Snap properly lights up 1 LED at a time as well.

alf45tar, 2024-11-10:
Use the latest version for sponsor. I re-enabled the led colors to use for Tap.

Pedalnetics, 2024-11-10:
That did it, fixed/was able to set the color and get it flashing.

closed 2024-11-10 by Pedalnetics

Pedalnetics, 2024-11-11:
Reopening to flag a bug in this: If the MIDI Clock is stopped while the Tap LED is lit, the TAP led stays lit solid and stops flashing.  I'd expect the proper outcome is when the MIDI Clock Off is sent, the Tap LED should be turned off.

alf45tar, 2024-11-11:
Fixed in the latest commit for version 3.5.1

Pedalnetics, 2024-11-11:
Confirmed, resolved via your commit