alf45tar/PedalinoMini #540: MIDI Clock BPM Sending different values than shown

Pedalnetics, 2024-11-11:
When running MIDI Clock, the BPM being shown on down-stream devices is not matching the Pedalino display.  I've connected to a Morningstar MIDI controller, a PirateMIDI Controller, as well as USB MIDI devices reading values via PocketMidi, and all concur with the findings.

First example, when first launching MIDI Master, the BPM is set to 120BPM but downstream always sees 125BPM.

Then the display is almost always 2-5 BPM off.  1 BPM I can see as a float-rounding error, but this seems to be beyond that.  I am waiting for at least 4 measures to ensure it's synced up as I noticed it seems to "walk up" to the value, so the photos are taken after the recieviing BPM have become steady/stopped changing.

For visual purposes, I've taken the photos with it feeding into a Morningstar MC6Pro as it has the BPM displayed easily to get into the same picture:
[image]
[image]
[image]
[image]
[image]

Pedalnetics, 2024-12-10:
@alf45tar any thoughts here?

zavorra, 2025-03-06:
ESP32 has an internal RTC but it’s not very accurate with frequency error of about 5%. I think than fixing that would require additional hardware

Pedalnetics, 2025-03-06:
The heltec devices have an external crystal on-board; it's also been predictable reproducable across multiple esp32-compatible device.