alf45tar/PedalinoMini #541: WS2812B pin - pedalino 6 build: wrong schematic?

A7F, 2024-12-06:
hi, just a quick question on the wiring shown [here](https://github.com/alf45tar/PedalinoMini-6/blob/main/images/PedalinoMini%206%20Wiring%201.jpg): you connect WS2812B DIN on pin D5 (GPIO 21). I am doing the exact same pedalino mini 6 build in the picture that is, with 6 momentary footswitch, therefore I suppose it would make more sense if I connect them to D1:D6 pins rather than A1:A6... this way, I think I would save up the pullup resistors... correct? However, by doing it this way I would need the D5 pin occupied by the LEDs. 
However, in [this other schematic](https://github.com/alf45tar/PedalinoMini/blob/master/images/Schematic_PedalinoMini_Sheet-2.svg) it looks like the strip should go on DATA pin, that is GPIO 03. I tried moving the strip there, but I had no control at all.
Any insight?

Thank you :)

alf45tar, 2024-12-09:
Pin assignment is critical because not all the pins are the same. I recently changed it because the previous one was not fully correct. Led strip on GPIO 21 is correct. The other schematic need to be updated.

A7F, 2024-12-13:
okay thank you :) so, if I got everything correct, in the case of pedalino mini 6 with those 6 temporary footswitches:
footswitch 1:6 to pin A1:A6 (gpio 11,12,13,17,19,20), led to gpio21
| what | ttgo physical | ttgo logical |
| ----- | --------------- | ------------- |
| footswitch 1 | gpio11 | A1 |
| footswitch 2 | gpio12 | A2 |
| footswitch 3 | gpio13 | A3 |
| footswitch 4 | gpio17 | A4 |
| footswitch 5 | gpio43 | D3 |
| footswitch 6 | gpio44 | D4 |
| ws2812b | gpio21 | DATA |
| trs 1 tip | gpio18? | A6 |
| trs 2 tip | gpio17? | A5 |
| trs 1 sleeve | gpio01? | D6 |
| trs 2 sleeve | gpio02? | D2 |

am I correct with the wiring? If yes, I would end up configuring the first 4 footswitches as "momentary", but what about switch 5 and 6?
Thank you, sorry for the stupid question but I prefer to check twice!

alf45tar, 2024-12-14:
Here the connections used for PedalinoMini 6.

Pedal|Button|Footswitch|GPIO
-----|-------|----------|------
1|1|FS1|11
1|2|FS2|44
2|1|FS3|12
2|2|FS4|43
3|1|FS5|13
3|2|FS6|10
5||TRS1|17, 2
6||TRS2|18, 1
|||LEDS|21

closed 2024-12-30 by alf45tar

A7F, 2025-01-07:
alright, I did it!
I designed the case pretty much as the pedalino mini 6, but with some changes to reduce print time. Here are some pictures:
[image]
LEDs are bright but colors can be recognized easily (picture doesn't make justice), as led diffuser I use 6mm round transparent rubber feets. Compared to pedalino mini 6, I also removed the FS numbers as they take too much print time, with older printers if not well calibrated they might turn out bad, and I thought labels are just easier overall, you can change them whenever you want to. Surface finish is from textured PEI plate, then I spray painted everything black.
[image]

and this is how I set up the controls, making sure to disable pedals 7 and 8 as are not used, so that they won't send any midi event at all:
[image]
with this setup, I still need to figure out how to set pedals with button 2 (such as FS2 in the wiring you sent) to do things like turning ON led 2... but that's another story :)

this is a look at the inside, I managed to fit 2x 18650 batteries wired in parallel for extra juice so that I can forget charging the pedalboard for quite some time...
[image]

I thought there was no need to 3d print the supports for the 18650 batteries, as in most cases their holder are provided for free with the battery itself. Then, I hot glued some components to hold them in place and... this is it :) very sturdy but super compact, I'm very happy how it came out and I printed it in like 4 hours (creality ender 3).
Let me know if you need the 3D model, I did it with fusion360 and if you allow me, I'll be happy to share it.