alf45tar/PedalinoMini #542: Please don't rely on `sysenv.TMPDIR` (It doesn't exist on all Linux distros)

T-vK, 2025-01-11:
I followed the build instructions, but I get an error during the pio build because when `TMPDIR` is not set, then the workspace dir gets set to `/PedalinoMini`. This is because of the line `workspace_dir = ${sysenv.TMPDIR}/PedalinoMini` in `platformio.ini`.
Please consider using a more robust mechanism that falls back to `/tmp`.

```
Can not remove temporary directory `/PedalinoMini/build`. Please remove it manually to avoid build issues
Processing bpi-leaf-s3 (platform: espressif32@6.9.0; framework: arduino; board: bpi_leaf_s3)
------------------------------------------------------------------------------------------------------------------------------------------------------
PermissionError: Traceback (most recent call last):
  File "/usr/lib/python3.12/pathlib.py", line 1311, in mkdir
    os.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/PedalinoMini/libdeps/bpi-leaf-s3'
...
```

I'm using Manjaro Linux btw. 

As a workaround for now, I had to change 
`workspace_dir = ${sysenv.TMPDIR}/PedalinoMini` 
to 
`workspace_dir = /tmp/PedalinoMini` 

... or build from the terminal instead of the GUI controls provided my pio/vscode with  
`TMPDIR=/tmp platformio run --environment bpi-leaf-s3`

zavorra, 2025-03-06:
+1 on that

T-vK, 2025-06-19:
Thanks - using `${sysenv.HOME}/.cache/PedalinoMini` would indeed be more reliable than `${sysenv.TMPDIR}`, and I totally see your point regarding the could syncing. 

Let's list the pros and cons for the options we'd have to make a good decision.

Why using `/tmp`, `$TMPDIR`  or any other temporary directories are not a good idea:
- Cross-Platform issues - Doesn't work in all environments (some don't have `/tmp`, some don't have `$TMPDIR` and some may lack both)
- Bad practice - Build artifacts are meant to persist to a certain degree to avoid unnecessary recompilation and dependency re-downloading
- Conflicts - If you maintain multiple copies or forks of the project or the system has more than one user account working on the project, you may run into conflicts because of the shared directory. This includes the fact that a `pio clean` will wipe the artifacts of all projects, slowing down compile times.

Why  `${sysenv.HOME}/.cache/PedalinoMini` would be a better solution:
- Better cross-platform support
- No conflicts in case of multiple user accounts working on the same project on the same system.

Why  `${sysenv.HOME}/.cache/PedalinoMini` is not perfect:
- Some environments set `$HOME` to `/` in which case you may run into permission issues because of creating directories directly in the root directory.
- You may still run into conflicts when working on multiple copies of the project with the same user account.
- There may be issues in environments where `~/.cache` doesn't exist by default.
- Cloud syncing may get triggered unnecessarily for unimportant build artifacts

Why not setting `workspace_dir` at all may be a better solution:
- Guaranteed cross-platform support
- No permission issues
- No conflicts when maintaining multiple copies or forks
- Following standard practices makes it easier for other developers to get familiar

I think to avoid unnecessary syncs, it would make sense to move those projects to a location that is excluded from cloud syncing entirely - after all, you already "sync" the important project files using git anyway and git by itself is smart enough to omit the `.pio` directory when configured correctly.

alf45tar, 2025-06-19:
Not sure why I decided to move outside the project folder. If I remember well is to implement the Build and Release GitHub Action.

T-vK, 2025-06-19:
I mean, you do have a couple of hardcoded paths like `/tmp/PedalinoMini/build/ttgo-t-eight/firmware.bin` in your workflow files, but if you were to remove the:
```
    env:
      TMPDIR: /tmp
```
and replaced the hardcoded `/tmp` occurrences with `./pio`, your Github Actions would work fine.

I've never had any issues with the default `.pio` directory in GitHub Actions - even in bigger projects.

If you ever feel like optimizing your Actions, there may be some interesting bits [in here](https://github.com/T-vK/ShowboxExpansionModule/blob/main/.github/workflows/build-and-release.yml) for you (like automatic versioning of the releases and directly reading the pio environments to build from the platformio.ini).

alf45tar, 2025-06-19:
Thanks for the suggestion. I will have a look to your code.

closed 2025-07-25 by alf45tar