alf45tar/PedalinoMini #550: Battery Level Indicator

regisampa, 2025-05-14:
Hello @alf45tar , I am using BPI Leaf S3 and my pedalboard is working perfectly except for the battery charge indicator, which keeps flashing as if it is charging and does not show the correct battery voltage.

My version is 3.5.0, how do I solve this problem? Has any member ever faced this? Thank you very much.

I connected a wire on the GPIO14 pin to the battery positive

#ifdef BATTERY
  #ifdef BATTERY_ADC_EN
    /*
      BATTERY_ADC_EN is the ADC detection enable port
      If the USB port is used for power supply, it is turned on by default.
      If it is powered by battery, it needs to be set to high level
    */
    pinMode(BATTERY_ADC_EN, OUTPUT);
    digitalWrite(BATTERY_ADC_EN, HIGH);
    delay(10);
  #endif //BATTERY_ADC_EN
    analogReadResolution(12);
    uint16_t voltage = analogReadMilliVolts(BATTERY_PIN) * 2; // Read the ADC voltage using the built in factory calibration, no conversion needed.
                                                              // Battery voltage is measured using a voltage divider
    analogReadResolution(ADC_RESOLUTION_BITS);
    #ifdef BATTERY_ADC_EN
    digitalWrite(BATTERY_ADC_EN, LOW);
    #endif //BATTERY_ADC_EN
    uint16_t difference = abs(voltage - batteryVoltage);
    //DPRINT("%d %d %d %d\n", voltage, batteryVoltage, difference, map2(constrain(batteryVoltage, 3000, 5000), 3000, 5000, 0, 100));
    if (difference > 100)
      batteryVoltage = (uint16_t)voltage;
    else
      batteryVoltage = (7 * batteryVoltage + voltage) / 8;
#endif // BATTERY

alf45tar, 2025-05-14:
I noted you uncommented `analogReadResolution(12);`. You should read the battery voltage now? What is the value of battery voltage in the web?

regisampa, 2025-05-14:
[image]

Is This?

alf45tar, 2025-05-15:
You should not connect battery positive to PIN 14. It is already connected by the board.

regisampa, 2025-05-15:
I actually connected the positive battery pin to GPIO14, I'm going to disconnect it. Does this solve the problem?

I'm powering the board through the 3.3V pin with the positive Battery, is this connection correct or should I power the board through the 5V pin? I've already jumped "in-out" on the board so that this pin works as a voltage input. Thank you very much.

alf45tar, 2025-05-16:
Check the board schematic. There is a specific connector for the battery which can recharge it when USB is connected. Pin 14 is connected to it.
The 3.3V pin is connected to a step-down DC-DC converter. Pin 14 will never read the battery value if battery is connected to 3.3V. It will always read the output of the step-down converter which is always 3.3V.

regisampa, 2025-05-16:
I see, so I think my board is not compatible for battery charge monitoring, as it does not have a specific connector for this purpose. My board is the ESP32-S3 N16R8 Chinese model. 

Is there any way to monitor the charge with this board model? Thank you very much for your support on this journey.

closed 2025-06-06 by regisampa